solokeys/solo1 #1: Finish NFC implementation

conorpp, 2018-09-13:


merlokk, 2018-10-27:
I know a guy who can help with antenna design
https://www.upwork.com/o/profiles/users/_~01ebbfa9e6b4a9119b/
If it needs)

manuel-domke, 2018-11-01:
I can also help with antenna design verification and testing.
As I'm part of your crowd already, feel free to reach out. I can at least support by providing some measurements for design verification.

acerspyro, 2019-02-05:
Apologies for the possibly dumb question, but does that mean I'll have to buy the hardware again once this is completed to get NFC support?

NFC does not appear to be working on my SoloKey, and I stumbled upon this issue.

conorpp, 2019-02-05:
Yes it is 2 different hardware models.  The NFC version has 2 main hardware changes:

* AS3956 for NFC communication
* Antenna/coil

The firmware is mostly done (thanks a lot to @merlokk) and the antenna design/tuning is almost done.

conorpp, 2019-02-05:
Please contact hello@solokeys.com for store/order related questions.

bbigras, 2019-02-14:
I was also under the impression that NFC was already supported. It would be a good idea to make it clear on https://shop.solokeys.com/ that it's not the case yet and mention if the current hardware will support it or not.

merlokk, 2019-02-14:
https://github.com/solokeys/solo/commits/nfc

conorpp, 2019-02-16:
@bbigras @acerspyro thanks for the feedback.  Definitely not our intent to mislead people on this.  We just updated our store to note current tokens don't support NFC.

conorpp, 2019-04-25:
:D

closed 2019-04-25 by conorpp