solokeys/solo1 #2: Decide on forum for development discussions

conorpp, 2018-09-13:
[Some suggestions](https://github.com/conorpp/u2f-zero/issues/76#issuecomment-420629143) that have been made:
- matrix.org
- Gitter
- Discord

I think setting up a Google group might be best for everyone, feel free to suggest otherwise though.

0x0ece, 2018-09-13:
Or Slack, Slack is cool :)

mvgorcum, 2018-09-13:
In the spirit of open source I'd argue for matrix or gitter.

I like matrix a lot, but I'm clearly biased.

MaPePeR, 2018-09-14:
Any reasons not to use GitHub?

Zjemm, 2018-09-14:
@mvgorcum  i agree, matrix is such a wonderful open project, as is this project here, so it fits nicely.

Also matrix growing rapidly in the opensource community, and is a great platform for real time discussions. and those live discussions you dont want to get on the github issue's page.

@MaPePeR Matrix can work together with github. so you can create issue's in the github from within matrix

nickray, 2018-09-15:
We probably won't have the resources to support this before and during the Kickstarter. I also think Matrix would fit quite nicely with our philosophy! Could even go super old school and cross link with an IRC channel :D However, I've seen the client be quite slow; perhaps this has improved.

The other issue is searchability of discussion history for newcomers to the project. A Google group might work, but that interface is quite cumbersome. For little things, I agree that Github issues are sufficient. For bigger things, perhaps a kind of "Solo Enhancement Proposal" process might make sense? With a design document sketching out a plan in advance?

We use Keybase for our internal communications, for now I suggest hopping onto https://keybase.io/team/solokeys.public if you'd like some direct communication.

nickray, 2018-10-07:
i reserved the solokeys matrix.org group and room in case we need it. this stuf is still slooow though

Zjemm, 2018-10-07:
Good, I'll join 😀

Let's go with matrix

nickray, 2018-10-07:
It is only reserved, not in use :) If you wanna chat, head to https://keybase.io/team/solokeys.public for now.

corbolais, 2018-10-18:
My feel is, keybase is not living up to FLOSS standards. W/o having dug into details, the codebase involves binary blobs. That's what's scaring me most.

Anyone fancing #solokeyssec @freenode? Waiting for *you* to /j #solokeyssec ;)

mvgorcum, 2018-10-18:
We can link the matrix room to the freenode room, wouldn't that be the best of both worlds?

corbolais, 2018-10-18:
dunno how matrix is operating..

sbrl, 2018-10-18:
Keybase has binary blobs? Oh dear.

I'd prefer XMPP myself instead of IRC/freenode, but I think I might be able to set up a personal gateway there if need be.

Krastanov, 2019-02-04:
Did a decision evolve? I am interested in a way to follow the bleeding edge development of the project, roadmaps, etc, but I have not been able to find any particularly recent forum.

corbolais, 2019-02-04:
Hi,

Some note dislike for keybase for various reasons.

In my behalf, I try to avoid using binary blobs. Then it's counterproductive to employ proprietary SW to promote user contribution.

Please use common methods for issue, bug, feature tracking and mailing lists, IRC channels for direct exchange.

Note aside: Does anyone know whether signal's group chat feature scales with increasing user count?

C.

On February 4, 2019 1:25:26 PM UTC, Starbeamrainbowlabs <notifications@github.com> wrote:
>@Krastanov: There's a Keybase group at the moment and a twitter
>account, along with a Discourse forum as far as I can tell.

stappersg, 2019-10-17:
(Breaking eight months of silence.)
Please prevent that this open issue will drain more human attention.

Decide on forum for development discussions

sbrl, 2019-10-17:
I mean I could host an XMPP multi-user chatroom on my server, but something else would probably be more suitable.

corbolais, 2019-10-17:
A bazillion of opensource projects use a thing called I R C. Maybe we could consider using something _like_ that, too?