solokeys/solo1 #4: Layout easy-to-solder board

conorpp, 2018-09-13:


corbolais, 2018-10-10:
hey conor,
thanks for your work on u2fa zero and the new project solo.

can help w/ schematics (cleanup etc.) and layout.

cheers

dkanada, 2018-10-11:
If you have svg files of the buton icon used on the housing I would also like to submit some alternative styles. It doesn't seem like the icon is used anywhere else yet but the letter looks a bit odd jutting out of the edge of the design in my opinion.

corbolais, 2018-10-11:
sry can't help you w/ svg files.

conorpp, 2018-10-12:
@corbolais Yeah totally!  I'll post Eagle schematic and layout details soon.

@dkanada yes, seeing some other styles would be great.  We still have time to update the case if we find something we like.  Attached is the SVG I used.

[logo.zip](https://github.com/SoloKeysSec/solo/files/2473518/logo.zip)

closed 2019-11-03 by 0x0ece