solokeys/solo1 #9: examples/credential.py fails

pjz, 2018-09-19:
Here's what I did:
```
pj@raft:~/src/solo$ ./venv/bin/python python-fido2/examples/credential.py

No FIDO device found
pj@raft:~/src/solo$ 
pj@raft:~/src/solo$ ./main &
[1] 30086
pj@raft:~/src/solo$ state file exists
init device
init ctaphid
init ctap
[STOR] Auth state is initialized
[STOR] pin not set.
recv'ing hid msg 

pj@raft:~/src/solo$ ./venv/bin/python python-fido2/examples/credential.py
Traceback (most recent call last):
  File "python-fido2/examples/credential.py", line 42, in <module>
    dev = next(CtapHidDevice.list_devices(), None)
  File "/home/pj/src/solo/venv/lib/python3.6/site-packages/fido2-0.3.1.dev0-py3.6.egg/fido2/hid.py", line 140, in list_devices
    yield cls(d, hidtransport.UsbHidTransport(dev))
  File "/home/pj/src/solo/venv/lib/python3.6/site-packages/fido2-0.3.1.dev0-py3.6.egg/fido2/_pyu2f/hidtransport.py", line 215, in __init__
    self.InternalInit()
  File "/home/pj/src/solo/venv/lib/python3.6/site-packages/fido2-0.3.1.dev0-py3.6.egg/fido2/_pyu2f/hidtransport.py", line 235, in InternalInit
    r = self.InternalExchange(UsbHidTransport.U2FHID_INIT, nonce)
  File "/home/pj/src/solo/venv/lib/python3.6/site-packages/fido2-0.3.1.dev0-py3.6.egg/fido2/_pyu2f/hidtransport.py", line 254, in InternalExchange
    ret_cmd, ret_payload = self.InternalRecv()
  File "/home/pj/src/solo/venv/lib/python3.6/site-packages/fido2-0.3.1.dev0-py3.6.egg/fido2/_pyu2f/hidtransport.py", line 308, in InternalRecv
    first_read = self.InternalReadFrame()
  File "/home/pj/src/solo/venv/lib/python3.6/site-packages/fido2-0.3.1.dev0-py3.6.egg/fido2/_pyu2f/hidtransport.py", line 300, in InternalReadFrame
    frame = self.hid_device.Read()
  File "/home/pj/src/solo/venv/lib/python3.6/site-packages/fido2-0.3.1.dev0-py3.6.egg/fido2/_pyu2f/test.py", line 40, in Read
    msg[i] = ord(v)
TypeError: ord() expected string of length 1, but int found
```

This is built and running on ubuntu18.04.

pjz, 2018-09-19:
Looks like maybe it's a py3.6 issue?  It worked when I changed test.py from

    for i,v in enumerate(pkt):
        msg[i] = ord(v)
    return msg
to

    for i,v in enumerate(pkt):
        msg[i] = v if type(v) == int else ord(v)
    return msg

Apparently in py3.6, indexing into a bytearray gets you an int, not a char. TIL.
With that fixed, I instead got:
```
pj@raft:~/src/solo$ ./venv/bin/python python-fido2/examples/credential.py

Touch your authenticator device now...

[CTAP] cbor input structure: 0 bytes
[CTAP] CTAP_GET_INFO
[CTAP] cbor output structure: 75 bytes
[CTAP] cbor input structure: 119 bytes
[CTAP] CTAP_MAKE_CREDENTIAL
[CTAP] cbor output structure: 917 bytes
New credential created!
CLIENT DATA: {"type": "webauthn.create", "clientExtensions": {}, "challenge": "Y2hhbGxlbmdl", "origin": "https://example.com"}
ATTESTATION OBJECT: AttestationObject(fmt: 'packed', auth_data: AuthenticatorData(rp_id_hash: h'a379a6f6eeafb9a55e378c118034e2751e682fab9f2d30ab13d2125586ce1947', flags: 0x45, counter: 452984832, credential_data: AttestedCredentialData(aaguid: h'00112233445566778899aabbccddeeff', credential_id: h'2ab51723da0582c44c4b0ad6e8d2867d00000000000000004eb142b2929864d52de84ade867dde2315bf393d59c46a7c7d7dd8e963d131461a2f93b9ff22a0bbf8229c76449d676f6cac74e3bc1d582e1ad77d1938b59a7f91d3390973c5148ef7f012d97bb20f12a90681c440cc7cb3bf2907ab830111918c91a97225ff752769c540b14e1113790a443d016c8d8d35fffd7df40c72d127e5134500cc3ffd48605d8de68f5cece3', public_key: {1: 2, 3: -7, -1: 1, -2: b'\x96\xd8V\xf2\x02<\xb4QV\xf4\xef\x96WT42<\x12s\xf6Q2i\xf4\xcb\xa8\xa6\x86\xb8\tu\x04', -3: b'{q\xec4=\x0b\xd6\x9f;\xaeD\x150\x83SW\xb0)\x93\xd8\xe8\x81\xbc\xdc\n\xe6\ti\xbe\xd4\xd7J'}), att_statement: {'alg': -7, 'sig': b'0F\x02!\x00\xaa\x12\xe1SN\xed\xaa\xb6Y}*\x91q\x908\xcb\xbbY\xb0\xd7\x96n{\xb2>~\x85y\x04\xce\xd8z\x02!\x00\xca\xf0\xe1\xca\xe9\x10\xa7\xa9+6,\x88\x93\xdab\x1d\x1ar\x11\xa4D]\xbe\x890[\xa3\xa5\xbc\x11\x9b\x7f', 'x5c': [b'0\x82\x01\xfb0\x82\x01\xa1\xa0\x03\x02\x01\x02\x02\x01\x000\n\x06\x08*\x86H\xce=\x04\x03\x020,1\x0b0\t\x06\x03U\x04\x06\x13\x02US1\x0b0\t\x06\x03U\x04\x08\x0c\x02MD1\x100\x0e\x06\x03U\x04\n\x0c\x07TEST CA0 \x17\r180510030620Z\x18\x0f20680427030620Z0|1\x0b0\t\x06\x03U\x04\x06\x13\x02US1\x0b0\t\x06\x03U\x04\x08\x0c\x02MD1\x0f0\r\x06\x03U\x04\x07\x0c\x06Laurel1\x150\x13\x06\x03U\x04\n\x0c\x0cTEST COMPANY1"0 \x06\x03U\x04\x0b\x0c\x19Authenticator Attestation1\x140\x12\x06\x03U\x04\x03\x0c\x0bconorpp.com0Y0\x13\x06\x07*\x86H\xce=\x02\x01\x06\x08*\x86H\xce=\x03\x01\x07\x03B\x00\x04E\xa9\x02\xc1.\x9c\n3\xfa>\x84PJ\xb8\x02\xdcM\xb9\xaf\x15\xb1\xb6:\xea\x8d?\x03\x03Ue}p?\xb4\x02\xa4\x97\xf4\x83\xb8\xa6\xf9<\xd0\x18\xad\x92\x0c\xb7\x8aZ>\x14H\x92\xef\x08\xf8\xca\xea\xfb2\xab \xa3b0`0F\x06\x03U\x1d#\x04?0=\xa10\xa4.0,1\x0b0\t\x06\x03U\x04\x06\x13\x02US1\x0b0\t\x06\x03U\x04\x08\x0c\x02MD1\x100\x0e\x06\x03U\x04\n\x0c\x07TEST CA\x82\t\x00\xf7\xc9\xec\x89\xf2c\x94\xd90\t\x06\x03U\x1d\x13\x04\x020\x000\x0b\x06\x03U\x1d\x0f\x04\x04\x03\x02\x04\xf00\n\x06\x08*\x86H\xce=\x04\x03\x02\x03H\x000E\x02 \x188\xb0E\x03i\xaa\xa7\xb78b\x01\xaf$\x97^~td\x1b\xa3{\xf7\xe6\xd3\xafy(\xdb\xdc\xa5\x88\x02!\x00\xcd\x06\xf1\xe3\xab\x16!\x8e\xd8\xc0\x14\xaf\tO[s\xef^\x9eK\xe75\xeb\xdd\x9bm\x8f}\xf3\xc4:\xd7']})

CREDENTIAL DATA: AttestedCredentialData(aaguid: h'00112233445566778899aabbccddeeff', credential_id: h'2ab51723da0582c44c4b0ad6e8d2867d00000000000000004eb142b2929864d52de84ade867dde2315bf393d59c46a7c7d7dd8e963d131461a2f93b9ff22a0bbf8229c76449d676f6cac74e3bc1d582e1ad77d1938b59a7f91d3390973c5148ef7f012d97bb20f12a90681c440cc7cb3bf2907ab830111918c91a97225ff752769c540b14e1113790a443d016c8d8d35fffd7df40c72d127e5134500cc3ffd48605d8de68f5cece3', public_key: {1: 2, 3: -7, -1: 1, -2: b'\x96\xd8V\xf2\x02<\xb4QV\xf4\xef\x96WT42<\x12s\xf6Q2i\xf4\xcb\xa8\xa6\x86\xb8\tu\x04', -3: b'{q\xec4=\x0b\xd6\x9f;\xaeD\x150\x83SW\xb0)\x93\xd8\xe8\x81\xbc\xdc\n\xe6\ti\xbe\xd4\xd7J'}
Attestation signature verified!

Touch your authenticator device now...

[CTAP] cbor input structure: 0 bytes
[CTAP] CTAP_GET_INFO
[CTAP] cbor output structure: 75 bytes
[CTAP] cbor input structure: 241 bytes
[CTAP] CTAP_GET_ASSERTION
[CTAP] cbor output structure: 332 bytes
Credential authenticated!
CLIENT DATA: {"type": "webauthn.get", "clientExtensions": {}, "challenge": "Q0hBTExFTkdF", "origin": "https://example.com"}

ASSERTION DATA: AssertionResponse(credential: {'type': 1, 'id': b'*\xb5\x17#\xda\x05\x82\xc4LK\n\xd6\xe8\xd2\x86}\x00\x00\x00\x00\x00\x00\x00\x00N\xb1B\xb2\x92\x98d\xd5-\xe8J\xde\x86}\xde#\x15\xbf9=Y\xc4j|}}\xd8\xe9c\xd11F\x1a/\x93\xb9\xff"\xa0\xbb\xf8"\x9cvD\x9dgol\xact\xe3\xbc\x1dX.\x1a\xd7}\x198\xb5\x9a\x7f\x91\xd39\ts\xc5\x14\x8e\xf7\xf0\x12\xd9{\xb2\x0f\x12\xa9\x06\x81\xc4@\xcc|\xb3\xbf)\x07\xab\x83\x01\x11\x91\x8c\x91\xa9r%\xffu\'i\xc5@\xb1N\x11\x13y\nD=\x01l\x8d\x8d5\xff\xfd}\xf4\x0cr\xd1\'\xe5\x13E\x00\xcc?\xfdH`]\x8d\xe6\x8f\\\xec\xe3'}, auth_data: AuthenticatorData(rp_id_hash: h'a379a6f6eeafb9a55e378c118034e2751e682fab9f2d30ab13d2125586ce1947', flags: 0x05, counter: 469762048), signature: h'3046022100dff80fc577b1bdc21a81c249665c87fe9d1b5498b34346b4665e23dd9a14e5dc022100d685e229c9c96917a291d65929860c435c11e5f32838b6b9cec09e9858dfa70a', user: {'id': b'user_id', 'displayName': 'A. User'}, number_of_credentials: 1)
Assertion signature verified!
```
...which I guess is correct?  The docs don't say what the output should be.

conorpp, 2018-09-19:
Yes this is a python 2 vs 3 error, and your fix looks perfect.  The output you see is correct.

Thanks for catching that!  I can commit your fix tomorrow sometime or can accept pull request :)

pjz, 2018-09-19:
I think it'll be easier to let you do it since it will involve a fix in your python-fido2 fork and also a fix in this repo as to what commit the submodule  should point to.

conorpp, 2018-09-19:
This was actually [already fixed](https://github.com/SoloKeysSec/python-fido2/commit/329434fdd476870ff0a73196b6de8a963409235f) in `python-fido2`, but I needed to update the submodule to point to latest commit.

closed 2018-09-19 by conorpp