solokeys/solo1 #12: BOOT0 pin/DFU?

hlandau, 2018-10-12:
In current documentation, it seems to be implied that a programmer will be required to reflash the device with user-provided code.

The STM32 series has a nice DFU bootloader in ROM. The ability to activate this from USB should rightly be a no-no for security reasons, but AIUI it's activated via the BOOT0 pin, which is currently unconnected.

The SWDIO/SWCLK pins already appear to be broken out, I assume to test pads, on the schematic. Any interest in breaking out BOOT0 as well so people without a STM32 programmer can boot it into DFU mode?

(Security considerations: None, since SWDIO/SWCLK are already present and enable reprogramming for anyone with physical access to the device. This enables someone to reprogram the device only if they have physical access, but without a programmer. Of course, the DFU application in boot ROM doesn't enable code readout protection to be bypassed; erasure must be performed first.)

nickray, 2018-10-13:
For regular users, Solo + Solo Tap will allow installing signed firmware updates (from our website, over USB). For advanced users, we will be offering a "Solo Hacker" that comes unlocked.

One thing we'd like to avoid is losing the on-device keys during the update process :)

hlandau, 2018-10-13:
Hmm, not sure I follow. Obviously, the DFU bootloader in ROM has no bearing on whether a flashed firmware image can implement functionality to allow signed updates under its own control.

What do you mean by 'coming unlocked'? AIUI, STM32s which ship unprogrammed can default into DFU mode, but once programmed you need the BOOT0 pin to get it back into it. A 'Hacker' version won't be too useful if one bad flashing prevents further reprogramming without having an STM32 programmer.

conorpp, 2018-10-13:
We're planning to have `level 2` readout protection enabled on normal Solo keys, so only our signed update bootloader can make changes.  For an 'unlocked' Solo, we plan to leave readout protection on `level 1`, so the DFU can be utilized.  

Rather than having people pull the BOOT0 pin a particular direction, we'll have a command in our bootloader that changes the ST chip internally to boot the DFU (see 2.6 in reference manual).  So the DFU can be activated via USB.  The DFU can later be told to boot the application when ready.  

That's what I'm thinking, open to other suggestions.

hlandau, 2018-10-16:
Hmm... can you explain what threat model you're trying to mitigate by using level 2?

For the 'unlocked' version, you refer to 'our' bootloader... do you intend to load a custom bootloader as the means for reprogramming? One of the advantages of the STM32s as opposed to e.g. the Microchip/Atmel SAMs is they have a ROM bootloader as opposed to a preloaded bootloader in Flash. There's a clear
security benefit here - for people who want to control their own roots of trust for firmware updates, they're likely to want to erase any preloaded code in mutable stores since it would be easy to compromise during shipping, so a flash-based bootloader isn't really helpful. Of course, it also reduces the amount of space available for the payload, however slightly.

The ROM-based DFU functionality is one of the things distinguishing the STM32s/this product in the first place, so it seems like a shame to waste it. Moreover, it would make the 'unlocked' SKU harder to brick.

conorpp, 2018-10-16:
With level 1, there's worry of some "coffee shop scenario" where someone briefly gets access to your key without you noticing and reflashes it with something malicious.  Sure, any secrets would be erased, but if someone didn't register any sites yet, it would go unnoticed.  Of course this is solved if people go through the verification themselves, but can't rely on consumers to do that.

We for each key, we plan to offer a "hacker version" that just has level 1 basically, for those that know what they're doing.

Having a flash-based signed-update mechanism still allows you to get new features since this project is continually being updated.  Also you can perform the updates from a website by leveraging U2F.

Also worth mentioning, keys purchased as level 2 can still be verified by leveraging attestation during registration step in FIDO2/U2F.  Basically there is a common secret on each device made by manufacturer, which is used for signature at registration, and can be verified with public key.

nickray, 2018-10-20:
Yeah we can't let consumer Solo keys be vulnerable to "BadUSB"-style attacks by leaving DFU open. For the "paranoid", I imagine purchasing "Solo Hacker", erasing everything, verifying "our" bootloader and initial choice of "our" firmware, perhaps compiling oneself (maybe in an easily used Docker container for "intermediate" users) and flashing that. Then you end up in the Solo ecosystem but with everything verifiable (except of course the hardware itself). Or, you can disable the signature check in "our" bootloader...

closed 2018-10-21 by nickray