solokeys/solo1 #14: feature discussion: touch button

corbolais, 2018-10-14:
Has there already been some thought about using a touch button (cap/res) instead of a push button?
To me it seems quite a nice substitution as it's one moving part less that can break/fail.

MaPePeR, 2018-10-14:
I asked this on Twitter some time ago: https://twitter.com/Blubb3rbub/status/1032644816386048000?s=19

So I also would prefer a touch button. Every time I read the description I think the "tactile button"-"feature" is actually quite the opposite

corbolais, 2018-10-16:
@conorpp hey, thanks for your feedback. I appreciate the time and effort you're putting into the project. That's a really good thing to see. You evaluated a variety of available button types, are the standard push buttons really that bad in terms of durability and reliability? Not that I'd like to trade the snap dome with a push button but .. ;)

BTW, I only now know of the aformentioned "discussion" on twitter.

I'm not just asking for you to implement it all on your own. I am sure willing to contribute, even more so when asking for a feature to be evaluated/implemented. So, if you were still open to see this through I'd do my best to come along. After all, I pledged for the KS campaign. :)

In this case, I so much prefer the light touch of the capacitive sensor to the "forceful" push button press. In my environment this almost always means to first catch the stick/dongle then hold it firmly and finally press the button completely. While it is feasible using both hands, it gets tedious and cumbersome trying to do it one handed. All the more when it comes to snap domes as in my xp they're even harder to press than normal push buttons.

So instead of reaching down/behind/wherever to grasp the thing and fumble to press a button that each time can be found in a different spot when you're not actually seeing the dongle, I envy yubikey 4 nano users where they only need to reach and fumble "in a general direction" to work it capacitively. I'm trying to describe a subjective commodity here.. ;)

I had the idea to add at least three solder pads or even better through holes/vias to be able to mod the dongle using a standard 3-pin (Vcc, GND, OUT) capacitive sensor module off ebay. There are nice ones out there, albeit most of them too clunky for the targeted dongle. So one can add a sensor to its liking. Then, it is only a hack. 

Better yet, integrate the logic directly, by OTOH increasing BOM and complexity. Not that the circuit is particularly hard or anything, it is just one more step to feature creep. (I shouldn't mention this when advocating the inclusion of a capacitive sensor.. ;-)

After all, when thinking about this, those 3-pin sensors should constitute a drop-in replacement for the snap dome/push button. Both require debouncing.

When it comes to the case to be refactored, capacitive touch sensors shouldn't be requiring direct surface access rather than relative proximity (actual distance subject to evaluate).  So maybe a soft indent is all we need and this alone would even improve haptic user xp.

Having said all this, do you already have some module/IC/solution in mind/at hand/to be chosen from?

So far so good, I'm interested in your thoughts.

cheers

conorpp, 2018-10-16:
All good points, yeah I think capacitive touch is a good option to have for sure, and is certainly easier to press.  Let's try it out.  I'm interested to see how well it works from a silicone case.

I wonder if the existing snap dome foot prints could be used?

[image]

Design is based on [STM32L442KC](https://www.st.com/en/microcontrollers/stm32l442kc.html), which has capacitive sensing channels.  It looks like there [needs to be an extra series capacitor](https://www.st.com/content/ccc/resource/technical/document/application_note/28/4b/40/7c/e6/68/44/9c/DM00087593.pdf/files/DM00087593.pdf/jcr:content/translations/en.DM00087593.pdf), not bad.

corbolais, 2018-10-16:
Wow, you got a sample there!? didn't know your design was that advanced. :)

Not sure about the proximity of the keyhole triggering FP. Reusing the snap dome footprint for the sense caps would most likely involve reshaping the footprint, I guess. From a glance the above sense cap pdf seems to suggest the mcu has basic cap sense support but that may actually be lightyears from a usable product. Do you have xp regarding this mcu and/or cap sensing with it? Does it work by means of pin mux? Would we need solder bridges for configuration (snap dome XOR cap sense)? Might it work to enable both of them at the same time, with some (marginal or not) spatial distance from each other? But then, if cap sense works sensibly (no pun intended), no one would ever need the snap dome any more as trying to actuate it the cap sense would have triggered long before.

*away studying the linked pdfs*

corbolais, 2018-10-16:
Did you notice [Tomu](https://www.crowdsupply.com/sutajio-kosagi/tomu/) ? Cortex M0 @25MHz.

It features the same form factor and same cap touch sensor position as the yubikey 4 nano which I happen to so much like.

Is there something on the bottom layer in the Solo design? Might there be any chance of shrinking it to a Solo Nano? 

So muchas ideas, so little time and space..

cheers

0x0ece, 2018-10-16:
We have the same issue internally... my role is to kill Conor's creativity to make sure we deliver the Kickstarter :D

conorpp, 2018-10-16:
I actually don't have much experience with the ST MCU (aside from reading the datasheet a lot, yes pins can be muxed), but I have messed with capacitive touch sensing some back in school.  My impression is it's not hard to implement, just need to have an ADC basically, but I could be missing something haha.  It seems the Tomu's Happy Gecko MCU doesn't have anything special for capacitive touch sensing.  

All things I'd like to try out! (after kickstarter units are delivered :D).

AtosNicoS, 2018-10-17:
Could it be possible to make both options work? So users could decide which button to use? Just another idea :)

frodomo, 2018-10-18:
Same here I would love to see a capacitive button :)

corbolais, 2018-10-19:
@conorpp While reading the STM32 datasheet (capacitive sensing part anyway) a big question mark hovered over the whole STM stuff. To use the capacitive sensing functionality, ST provides a lib. In itself, nice thing. But is it FOSS? Or just another shared object to use w/o any way to audit and build it on my build box? How much so is this valid for the entire project? I haven't downloaded the STM SDK to look after it myself yet. If you're using it already, could you tell what's in the SDK bag'o'surprises today?

thanks
c

corbolais, 2018-10-19:
About TSC:
- "Designed to operate with STMTouch touch sensing firmware library". This part: "firmware library" does not sound very promising. Though it may mean nothing at all..
- "Full hardware management of the charge transfer acquisition sequence" sounds easy-peasy. All-inclusive-MCU, nice. Only the sw part to go. Well, we need some sensing area.. Could you hack one of your prototype pcbs to get some proof of concept xps?
- "Capacitive sensing technology is able to detect finger presence near an electrode which is protected from direct touch by a dielectric (glass, plastic, ...)". This seems even better than yubikey 4 nano's solution. I think, you're to touch the thing to make it do. So, in this "case", it's gonna work? (-> prototype testing)

If you got some spare pcbs, parts (can smd solder it) and sw or ready-to-use prototypes I can do some testing in this area. Need to get ahold of some  PPS or NPO caps then.

Whaddyathink?

conorpp, 2018-10-19:
The ST SDK and libraries should all be open source, no magic binaries.  They have ST's own version though, which is just typically to absolve ST of any liability.

Hmm those points all sound promising!  I'm interested to see how well it works for sure.

I can solder a sample and mail it to you sometime this weekend!

corbolais, 2018-10-29:
@conorpp Hey, just wanted to query an update on the issue. What's the plan? :)

stappersg, 2019-10-18:
(a year past by)

The touch button got implement in the next generation hardware known as [somu](https://solokeys.com/somu).

Because the `README.md` does mention  _somu_, do I think that it is OK to close this issue.

corbolais, 2019-10-18:
This was meant to be implemented by the solo, not the somu. I do not intend to by a whole new product just to get the cap touch feature. It is possible to mod the solo, this is what this thread is about.

Therefore I do disagree with the intend to close this.

@conorpp do you still pursue your intent to mail a sample in?