solokeys/solo1 #15: software-feature: primary u2f-token invalidating backup

corbolais, 2018-10-18:
Here's a good way to solve the [u2f-backup-token problem](https://dmitryfrank.com/articles/backup_u2f_token).

He even forked and [patched the u2f-zero repo](https://github.com/dimonomid/u2f-zero). So, once there's a PR, it would be easy to expand u2f-zero/solo usefulness.

RFC.

cheers

MaPePeR, 2018-10-18:
There is nothing that prevents an attacker that got physical access to your key to reflash a new firmware that has an even higher boost value than the backup key to re-enable that key, though?

But that is not the main concern here anyway, i guess.

corbolais, 2018-10-18:
> There is nothing that prevents an attacker that got physical access to your key to reflash a new firmware that has an even higher boost value than the backup key to re-enable that key, though?

I guess.

> But that is not the main concern here anyway, i guess.

I guess not.

 :)

vielhuber, 2019-10-31:
Hi! is there any update on implementing this idea? I will start playing around with u2f zero until then.

merlokk, 2019-10-31:
You can just modify firmware and add command to load/save whole memory .
but it totally insecure....

maybe needs to create something like key sharing system like in Gemalto SafeNet HSM, it can be done but it very hard to do for small amount of money)

vielhuber, 2019-10-31:
It seems that the only U2F device with backup functionality is https://trezor.io/. 
Infos about U2F: https://wiki.trezor.io/U2F
Infos about Backup: https://wiki.trezor.io/Recovery_seed
Are there any other devices?