solokeys/solo1 #16: OpenPGP support?

nimbius, 2018-10-18:
any chance getting openpgp support added?

nickray, 2018-10-19:
Hi @nimbius, yes extendability is a major goal of this project, thank you for asking.
We will communicate a public roadmap after the Kickstarter is over, but adding support for GPG, SSH, static passwords and more will definitely be part of it :)

binaryplease, 2018-12-21:
> Hi @nimbius, yes extendability is a major goal of this project, thank you for asking.
> We will communicate a public roadmap after the Kickstarter is over, but adding support for GPG, SSH, static passwords and more will definitely be part of it :)

Is this "public roadmap" out yet? Any progress on GPG/SSH? I would be very interested in seeing a GPG and SSH implementation. Actually this would be the main thing I would like to use this key for. Having my private keys not leaving the device would be great.

pjz, 2019-02-19:
Related: http://www.fsij.org/category/gnuk.html - runs on a STM32F103, but doesn't do FIDO or etc, only OpenPGP card protocol version 2.

michaelblyons, 2019-05-23:
For anyone lurking here who hadn't noticed yet, you may be interested in [`solokeys/openpgp`](https://github.com/solokeys/openpgp).

lopezator, 2019-06-26:
@michaelblyons sorry for the silly question, but this need the hacker edition I guess?

manuel-domke, 2019-06-26:
@lopezator:  fYes, a Solo Hacker is required to test and develop that firmware version.

At least as long as there is no officially signed firmware for it, which will probably only happen when it's stable.

manuel-domke, 2019-06-26:
@lopezator:  Yes, a Solo Hacker is required to test and develop that firmware version.

At least as long as there is no officially signed firmware for it, which will probably only happen when it's stable.

sakoula, 2019-08-13:
hi, is there any progress on this? So currently there is no way to store gpg or ssh keys on solo, right?

Frederick888, 2019-08-13:
@sakoula You can, but only on Hacker versions if I understand correctly.

sakoula, 2019-08-13:
@Frederick888 Thanks! I just have the regular version. 

I recall that when I signed up for the kickstarter one of the 'selling' points was support for gpg/ssh keys. Any idea if there are any plans of a soon-to-released firmware supporting these features?

Thanks!

janisz, 2019-08-13:
I think they need more money and try to collect it on https://www.crowdsupply.com/solokeys/somu#stretch-goals
> $100k - SSH/GPG - we’ll invest part of the funding to speed up the implementation of SSH/GPG support, starting with ECDSA keys.

conorpp, 2019-08-13:
We are still working on it here:
https://github.com/solokeys/openpgp

Still have some testing and porting to do.  If you want it faster, [consider backing our Somu campaign](https://www.crowdsupply.com/solokeys/somu#stretch-goals), which will continue to fund this effort!  And you get another cool key :)

Frederick888, 2019-08-13:
> https://www.crowdsupply.com/solokeys/somu#stretch-goals

Hmm... I've already got a USB-C Solo Tap and at least for now I haven't got a need for a mini security key (tbh I don't think leaving your security key plugged into your computer is a good practice). I would probably consider backing the campaign if there's a Tap Hacker option but that would lead to effectively less fund for Somu I guess (a little off-topic...).

lopezator, 2019-08-13:
@Frederick888 you can configure & store in a safe a Somu as a backup in case you lose your main key, that indeed would be a food practice.

lf94, 2020-10-22:
So uh, what happened?

lf94, 2020-10-29:
@conorpp ?

wucke13, 2021-01-26:
This for sure is not the right place. but I want to state this somewhere: I bought the original SoloKey, and the Somu. I convinced friends to do so as well. I will, however, not buy any more Solo stuff as long as this issue can not be closed. Make it a real alternative to YubiKey!

nimbius, 2021-01-26:
Solov2 just released (I funded it this morning) and surprise, no mention of pgp. :/

This is not a terribly difficult undertaking.  I hope to see more activity in this effort in 2021 and will eagerly test and debug any firmware offered for the v2.

nickray, 2021-01-26:
Yes, the appropriate place would be https://github.com/solokeys/kickstarter2021/discussions.

06kellyjac, 2021-01-26:
Or if you fund the Solo v2 you can fill out the survey (you'll receive a message about it) and mention this issue towards the end

dudeisbrendan03, 2021-05-17:
Any chances of OpenGPG support being introduced to the Solo 1?

maweil, 2021-06-03:
How about voting which gpg use cases would be the most important and implementing a small subset of the functionality only?
If the hardware of Solo v1 is not sufficient for a GnuPG Smartcard Implementation, exposing some vendor commands for signing or decrypting might be an option.

I'd be really happy to see at least the signing capabilities written by @rgerganov in PR #397 in the official firmware.

larcenou, 2024-03-05:
> Hi @nimbius, yes extendability is a major goal of this project, thank you for asking. We will communicate a public roadmap after the Kickstarter is over, but adding support for GPG, SSH, static passwords and more will definitely be part of it :)

I bought my first solo 5 years ago - the promise of active work on pgp implementation was a major reason for me to buy several more over the years. Today I have to admit it’s getting really disappointing to read the same phrase in the FAQ and still no pgp update rolled out.

kevinbarbour, 2024-03-05:
I agree, I also purchased my Solokey under the impression that PGP support was right around the corner. It's been a very disappointing product with the lack of effort they've put into achieving these early roadmap goals.

jans23, 2024-03-06:
For those interested in an alternative, our [Nitrokey 3](https://www.nitrokey.com/products/nitrokeys) has full OpenPGP Card support. It's also based on Trussed, the same firmware framework Solokey is using (and written in Rust).

Scrumplex, 2024-03-06:
Nitrokey was also on my radar as an alternative to Solokeys. I wonder if people made custom cases for the Nitrokey 3, because I personally think they don't *look* as appealing as a SoloKeys key.

larcenou, 2024-03-06:
> For those interested in an alternative, our [Nitrokey 3](https://www.nitrokey.com/products/nitrokeys) has full OpenPGP Card support. It's also based on Trussed, the same firmware framework Solokey is using (and written in Rust).

Does the Nitrokey 3c NFC also support using PGP via NFC? If yes I would immediately order at least 3 of them.

lf94, 2024-03-06:
@jans23 
> For those interested in an alternative, our [Nitrokey 3](https://www.nitrokey.com/products/nitrokeys) has full OpenPGP Card support. It's also based on Trussed, the same firmware framework Solokey is using (and written in Rust).

If we have proof of purchase of SoloKey v1s can we get a free Nitrokey? :slightly_smiling_face: 

I spent about $100 to get 3 SoloKeys just to have them be pretty much useless.

I would be an instant Nitrokey evangelist, praising.

jans23, 2024-03-06:
OpenPGP Card works via USB only, not NFC.

I'm not aware of any custom made casings but feel free to do so.

larcenou, 2024-03-06:
on my very old Yubikey PGP does work via NFC

nimbius, 2024-03-31:
solo2 is out and the roadmap looks like PIV is the teams primary focus :-1:

nimbius, 2024-03-31:
solo2 is out and the roadmap looks like PIV is the teams primary focus 👎

closed 2024-03-31 by nimbius

michaelblyons, 2024-03-31:
Change close type to Not Planned?