solokeys/solo1 #19: Compiling

nickray, 2018-10-20:
I started work on compiling the firmware on Linux, documented in <https://solo.solokeys.io/building>.

We need a dual Win/Linux compilation story (plus, the firmware gets stuck on the Nucleo 😜).

nickray, 2018-10-21:
And shouldn't we include debug symbols? AFAIK they don't increase size in flash.

Immortalin, 2018-10-21:
Why don't they increase size?

nickray, 2018-10-21:
> Why don't they increase size?

There's no use for them on the MCU, so they don't get loaded. I can't seem to find a good reference for this - we should probably collect some "basics of embedded gdb/openocd flashing+debugging" somehwere. But you can for instance compare `arm-none-eabi-size /path/to/your.elf --format=SysV` with  the output of `load` in `gdb` when flashing an ELF containing `debug_info`.

gaetanww, 2018-11-22:
I read this in the doc:
> Note that our python-fido2 fork will only connect to the software FIDO2 application, not a hardware authenticator. Install Yubico's fork to do that.
Does that mean that solo can be compiled to a virtual software authenticator?

szszszsz, 2019-01-05:
Hi @nickray! You have mentioned in https://github.com/SoloKeysSec/solo/issues/19#issue-372157989, that firmware got stuck in your case. Could you elaborate?

nickray, 2019-01-05:
I am afraid I don't remember. Word of warning though, Linux builds currently don't work properly (they build, but flashing them has issues), only Windows ones. I am hoping @conorpp comes up with a fix for that soon!

nickray, 2019-01-05:
Do you have a specific issue @szszszsz?

szszszsz, 2019-01-07:
@nickray I was trying then to run the firmware on Nucleo board, and could not identify the problem. It was fixed 2 days ago by @conorpp, thank you! 

Now I hope to connect to its serial port, and look inside. Have you tested this under Linux? Or do you work only on Windows with it?

yparitcher, 2019-01-07:
i have successfully built and flashed the firmware on Linux / Ubuntu

conorpp, 2019-01-07:
Serial port works fine on Linux but I had some problems at first.  Since it looks like a modem, the linux daemon `ModemManager` will [automatically start messing with it](http://www.chibios.com/forum/viewtopic.php?t=3057).  I had to disable it:

```
systemctl stop ModemManager 
```

Might be something that can be changed in the USB descriptors to prevent this from happening, opening #62 for it

szszszsz, 2019-01-08:
@conorpp It looks like that was the cause in my case as well (Fedora 29, Gnome 3) - I have disabled that service entirely. Thank you for the tip! Works as advertised now.

yparitcher, 2019-01-08:
> Serial port works fine on Linux but I had some problems at first. Since it looks like a modem, the linux daemon `ModemManager` will [automatically start messing with [...]

there is no need to disable ModemManager, just add the folowing to your udev rules and ModemManager will ignore it. (taken from ModemManager docs)
```
ACTION!="add|change|move", GOTO="mm_usb_device_blacklist_end"
SUBSYSTEM!="usb", GOTO="mm_usb_device_blacklist_end"
ENV{DEVTYPE}!="usb_device",  GOTO="mm_usb_device_blacklist_end"

ATTRS{idVendor}=="0483", ATTRS{idProduct}=="a2ca", ENV{ID_MM_DEVICE_IGNORE}="1"

LABEL="mm_usb_device_blacklist_end"
```

nickray, 2019-03-22:
> Why don't they increase size?

A "kind of" reference: https://mcuoneclipse.com/2012/06/01/debug-vs-release/

closed 2019-11-03 by 0x0ece