solokeys/solo1 #21: Continuous Integration

nickray, 2018-10-21:
Opening a ticket for discussion/input: [good read](https://jamesmunns.com/blog/hardware-ci-overview)

pjz, 2018-10-21:
This is such a general topic I think you should maybe use this ticket to discuss/consider/enumerate explicit CI features that are desired. General areas for research/inclusion are:

  * Code formatting: Black for python (see #22 ) , but anything for C? Even the linux kernel has a preferred formatting script.  
  * Static analysis/linting - 
    * python
       * ??
    * C
      * valgrind ?
      * cppcheck ?
      * There's a whole list at https://en.wikipedia.org/wiki/List_of_tools_for_static_code_analysis#C,_C++ that should be gone through to see what might be useful
      
  * test framework
    * python - most python projects use pytest, which is great, and we should set up the repo to have a clear place for tests to go and the CI to invoke those tests
    * C - there are a few unit test frameworks, but someone needs to choose one and set up the repo to have a clear place for them and the CI to get them built and run

0x0ece, 2018-10-25:
hey, I run `black` on `tools/`. It changed everything :) Then I run both `flake8` and `pycodestyle` and both were still complaining.

How about we setup pytest clean, and we start using `flake8` 100% there, and when black is ready we retest it?

0x0ece, 2018-10-25:
On the C side, I quickly tried:
- cppcheck, that resulted in https://github.com/SoloKeysSec/solo/pull/25
- oclint
- uncrustify

For the last 2... I couldn't really run them successfully. I think one way to go would be to setup CMake (maybe not as a main build system, but at least as a parallel one), as both sort-of rely on CMake architecture/structure to import all the files in a complex project.

We could start by integrating cppcheck, and go from there.

0x0ece, 2018-10-25:
On valgrind, I'm having issues on the Mac... I think because of the Mac, not because of Solo.

In https://github.com/SoloKeysSec/solo/pull/24 I'm proposing to catch sigint on the PC implementation, so we can cleanup properly. Once main is built, we could run valgrind main, run a bunch of tests, then kill main, and see the results from valgrind.

This, however, only tests fido2 app + pc impl. It doesn't really test the device-specific code.

ignaloidas, 2019-01-17:
@0x0ece `flake8` and `black` have different opinions and it's not likely that they will become compatible any time soon. So there is no option to use `flake8` and then switch to `black`. I personally find `black`'s code style easier to read than `flake8`'s.

0x0ece, 2019-01-17:
Black it is then!

yparitcher, 2019-01-17:
in my pull request #51 i started this by integrating black, and by making Travis build each solo build configuration with -Wall -Werror to see if there are errors and warnings, similar to what @0x0ece did in #24 for the PC build, this has helped clean up the code, any more integrations would be better, see #69 by @pjz

closed 2019-11-03 by 0x0ece