solokeys/solo1 #22: Use Black?

nickray, 2018-10-21:
Planned languages for the project are C/Assembler/Rust2018 (for firmware) and Python (3.6+, for scripting).
Rust has `rustfmt`, C is a... mess, but Python now has [black](https://github.com/ambv/black).

Is there any opposition to using it? Should we allow `--skip-string-normalization`?

Issue will be closed if it devolves into too much bikeshedding 😁

pjz, 2018-10-21:
Never heard of it, but if you want to get all B&D on code formatting it's as good a choice as any.  PEP8 leaves a lot of leeway, Black looks to remove most (if not all ) of it.

pjz, 2019-01-22:
Answers pointed to yes, since that's what in the Makefile in `master`.  So I added it to be run by CI in #86.

closed 2019-01-22 by nickray

pjz, 2019-02-06:
@nickray merge #86 ?