solokeys/solo1 #26: Add instructions on how to setup U2F and UAF on Linux desktops and servers

nickray, 2018-10-27:
Perhaps provide some scripts?

nickray, 2018-10-27:
prior art: https://medium.com/antelabs/using-u2f-at-linux-login-25b77d005e30

Espionage724, 2019-01-17:
For a quick how-to on logging in with Fedora 29 and GNOME:

- Create key:

`pamu2fcfg > ~/'u2f_keys'`

- Move it to Centralized:

`sudo mv ~/'.config/Solo/u2f_keys' '/etc/u2f_mappings'`

- Edit GDM so that it can use U2F:

`sudo -e '/etc/pam.d/gdm-password'`

- ...by adding the following line under the pam_selinux_permit line:

`auth        required	  pam_u2f.so authfile=/etc/u2f_mappings interactive cue`

- ...so that it looks like this:

```
auth     [success=done ignore=ignore default=bad] pam_selinux_permit.so
auth        required	  pam_u2f.so authfile=/etc/u2f_mappings interactive cue
auth        substack	  password-auth
auth        optional	  pam_gnome_keyring.so
auth        include	  postlogin
```
This will force you to have to have the key to log-in. This also prompts for your password too still, which also auto-unlocks the GNOME Keyring upon logging in.

You can position the pam_u2f.so line somewhere else and use `sufficient` instead of `required` so that you can log-in only with using that (no password prompt), but if you have anything using GNOME Keyring on log-in, you'll be prompted for the password for that. You can get around that by not using a password for the keyring (and thus, reset the keyring), but this leaves passwords in plain-text. The Keyring doesn't seem able to use PAM for authentication.

rafaelmr2, 2019-03-18:
When I try to run `pamu2fcfg` on Fedora 29 I get:

`Unable to generate registration challenge, error in transport layer (-2)`

Thanks

rafaelmr2, 2019-03-18:
I rebooted now and getting the same error. I install pam-u2f too. It's working normal in other web services.

Espionage724, 2019-03-18:
I'd try adding `--debug` and see if there's anything specific that might be shown:

```
pamu2fcfg --debug > ~/'u2f_keys'
```

rafaelmr2, 2019-03-19:
```
USB send: 00ffffffff8600080807060504030201000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
USB write returned 65
now trying with timeout 2
now trying with timeout 4
USB read rc read 64
USB recv: ffffffff860011080706050403020102000000020000000500000000000000000000000000000000000000000000000000000000000000000000000000000000
device /dev/hidraw3 discovered as 'Solo 2.0.0'
  version (Interface, Major, Minor, Build): 2, 0, 0, 0  capFlags: 5
JSON: { "challenge": "uydioPCOv5KPe9Unz3PUEjEbuDqGq1yZ1HWi9hXIcts", "version": "U2F_V2", "appId": "pam:\/\/dssc-fd3ba9.id.ufrj.br" }
JSON challenge URL-B64: uydioPCOv5KPe9Unz3PUEjEbuDqGq1yZ1HWi9hXIcts
client data: { "challenge": "uydioPCOv5KPe9Unz3PUEjEbuDqGq1yZ1HWi9hXIcts", "origin": "pam:\/\/dssc-fd3ba9.id.ufrj.br", "typ": "navigator.id.finishEnrollment" }
JSON: { "challenge": "uydioPCOv5KPe9Unz3PUEjEbuDqGq1yZ1HWi9hXIcts", "version": "U2F_V2", "appId": "pam:\/\/dssc-fd3ba9.id.ufrj.br" }
JSON app_id pam://dssc-fd3ba9.id.ufrj.br
USB send: 0002000000830049000103000000403d167575f1cfc54b5926fe0b55a5a7dc83c864ee68331919a46d0ef48671c8d8088785231627f0f40bf01f619f9b3556b1
USB write returned 65
USB send: 000200000000eeddd67b090c82f7e8b8bd76c5fd0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
USB write returned 65
now trying with timeout 2
now trying with timeout 4
now trying with timeout 8
now trying with timeout 16
now trying with timeout 32
now trying with timeout 64
now trying with timeout 128
now trying with timeout 256
now trying with timeout 512
now trying with timeout 1024
now trying with timeout 2048
now trying with timeout 4096
USB read rc read 64
USB rc -2
Unable to generate registration challenge, error in transport layer (-2)
```

conorpp, 2019-03-19:
It looks like it sends a registration request and doesn't get anything back after.  Does the LED on the key change to orange?  Do you press the button?

rafaelmr2, 2019-03-19:
I was forgetting to press the button, it works! :D

closed 2019-11-03 by 0x0ece

benemanuel, 2020-03-16:
> For a how-to on logging in with Fedora 29 and GNOME:
> 
>     * Install packages:
> 
> 
> ```
> sudo dnf install pamu2fcfg pam-u2f
> ```
> 
>     * Create key (make sure the Solo key is plugged-in and be prepared to press it when prompted):
> 
> 
> ```
> pamu2fcfg > ~/'u2f_keys'
> ```
> 
>     * Move it to Centralized:
> 
> 
> ```
> sudo mv ~/'.config/Solo/u2f_keys' '/etc/u2f_mappings'
> ```
> 
>     * For using Solo to authenticate through GDM (log-in and screen unlock), edit this file:
> 
> 
> ```
> sudo -e '/etc/pam.d/gdm-password'
> ```
> 
>     * ...by adding the following line under the `pam_selinux_permit.so` line:
> 
> 
> ```
> auth sufficient pam_u2f.so authfile=/etc/u2f_mappings cue
> ```
> 
>     * ...so that `/etc/pam.d/gdm-password` looks something like this:
> 
> 
> ```
> auth     [success=done ignore=ignore default=bad] pam_selinux_permit.so
> auth sufficient pam_u2f.so authfile=/etc/u2f_mappings cue
> auth        substack	  password-auth
> auth        optional	  pam_gnome_keyring.so
> auth        include	  postlogin
> 
> account     required	  pam_nologin.so
> ...
> ```
> 
>     * And for using Solo to authenticate `sudo`, edit this file:
> 
> 
> ```
> sudo -e '/etc/pam.d/sudo'
> ```
> 
>     * ...and add this under the `#%PAM-1.0` line:
> 
> 
> ```
> auth sufficient pam_u2f.so authfile=/etc/u2f_mappings cue
> ```
> 
>     * ...so that `/etc/pam.d/sudo` looks something like this:
> 
> 
> ```
> #%PAM-1.0
> auth sufficient pam_u2f.so authfile=/etc/u2f_mappings cue
> auth	   include	system-auth
> account    include	system-auth
> password   include	system-auth
> ...
> ```
> 
> If the Solo key is inserted, both GDM and sudo will prompt you (cue) to touch the device to authenticate. If the key is touched, you'll authenticate (no traditional password prompt). If the key isn't touched in a certain amount of time, or if the key isn't inserted, it will fallback to the traditional password entry authentication. The Solo key isn't required, but makes authenticating more convenient in this case.
> 
> The above uses `sufficient` to not require the Solo key, but `sufficient` can be changed to `required` if you want to require **both** the Solo key and traditional password entry. If the Solo key isn't inserted, authentication will fail without mention about the key, regardless if the correct password is entered.

worked perfectly.

nickray, 2020-03-17:
Hey @benemanuel, thanks for the update, any chance you could add your write-up as a pull request against https://github.com/solokeys/solo/tree/master/docs/solo ? That way it would turn up under https://docs.solokeys.io/solo and be way more discoverable. If so, feel free to add a new page, we can always reorganize later.

benemanuel, 2020-03-17:
I am lost on how to do this.

nickray, 2020-03-17:
Docs on contributing docs exist, but are sparse: https://docs.solokeys.io/solo/documenting/

You'd fork this repository, add your file and send a pull request.

If Python isn't your thing, can you convert your comment to a GitHub gist in Markdown format? Then I can commit that for you.