solokeys/solo1 #30: Extension and firmware upgrade

lcw99, 2018-11-20:
The current firmware upgrage and extension depend on the U2F API. From my knowledge, U2F is an API supported only in Chrome. In CTAP2 supported by all major browsers, it is not easy to transmit arbitrary extension data with web browser api. If I'm right, is there a way to solve this problem?

merlokk, 2019-02-13:
#104

conorpp, 2019-02-13:
CTAP2 API can still be used, and if a token only supports CTAP1/U2F, the browser will map everything to CTAP1 if it supports U2F.  Chrome, Firefox, and Edge all support CTAP1 so it should work.  I think Edge has some problems in that it always requires a pin to be set.

nickray, 2019-02-13:
Branch `u2f-over-webauthn` of https://github.com/solokeys/solo-station uses this WebAuthn with CTAP1 fallback. Indeed the hope is that it will eventually work on Edge + Safari too. So there is no U2F API in use anymore. A disadvantage is that you get constant popups when you flash the firmware upgrade :/

Regarding official support for CTAP2 extensions in WebAuthn, I doubt it will happen (soon), as it's implemented as extension of the Credential interface and only exposes create + get, just with different options than for passwords. We could petition the standards body to add some kind of `credential.custom` exposing CTAP2 :)

closed 2019-11-03 by 0x0ece