solokeys/solo1 #31: Virtual authenticator?

gaetanww, 2018-11-22:
Reading through the docs and especially (Readme.md):
> Note that our python-fido2 fork will only connect to the software FIDO2 application, not a hardware authenticator. Install Yubico's fork to do that.

I understood that it was possible to build solo as a software virtual authenticator? Is that true? If so, what is the status on that?

Thank you for your help

gaetanww, 2018-11-22:
I did some more digging around, so I'm goin g to answer part of my question:
- This project is supposed to build a "virtual authenticator" that interact with python-fido2 through udp. However, it is not currently building, see [Travis](https://travis-ci.com/SoloKeysSec/solo/builds)
I found the last commit that built successfully: [Travis](https://travis-ci.com/SoloKeysSec/solo/builds/89104010) and built it without problem.

gaetanww, 2018-11-22:
Yes that is what I meant by "virtual authenticator", sorry for the confusing terminology. 
It's just that `make all` didn't work on HEAD, because some functions aren't implemented in pc/device.c, so I was confused as to whether or not it was possible to run solo on linux (over UDP). 

(just to clarify, I used "virtual authenticator" because that is what google [called it](https://github.com/google/u2f-ref-code#a-virtual-software-u2f-device) for their u2f implementation)

conorpp, 2018-11-23:
I added the missing functionality to `pc/device.c` and tested it works now (#33).  It can be used as a software-only/virtual authenticator.  It doesn't check for user presence and stores all state in local files.  The counter and resident keys aren't stored, but that is relatively easy to add if needed.

gaetanww, 2018-11-23:
Awesome, thanks!

closed 2018-11-23 by gaetanww