solokeys/solo1 #36: Compiler target STM32L442

tiena2cva, 2018-11-29:
I try to compile solo firmware for STM32L442 (under target/stm32l442):
```bash
cd targets/stm32l442
make all
```
But I got error:
```
arm-none-eabi-gcc src/main.o src/init.o src/redirect.o src/flash.o src/rng.o src/led.o src/device.o src/fifo.o src/crypto.o src/attestation.o src/startup_stm32l432xx.o src/system_stm32l4xx.o lib/stm32l4xx_ll_rng.o lib/stm32l4xx_ll_gpio.o lib/stm32l4xx_hal_pcd.o lib/stm32l4xx_ll_usb.o lib/stm32l4xx_hal_pcd_ex.o lib/stm32l4xx_ll_pwr.o lib/stm32l4xx_ll_rcc.o lib/stm32l4xx_ll_tim.o lib/stm32l4xx_ll_usart.o lib/stm32l4xx_ll_utils.o lib/usbd/usbd_conf.o lib/usbd/usbd_core.o lib/usbd/usbd_ioreq.o lib/usbd/usbd_hid.o lib/usbd/usbd_desc.o lib/usbd/usbd_ctlreq.o ../../fido2/util.o ../../fido2/u2f.o ../../fido2/test_power.o ../../fido2/stubs.o ../../fido2/log.o ../../fido2/ctaphid.o ../../fido2/ctap.o ../../fido2/ctap_parse.o ../../fido2/main.o ../../crypto/sha256/sha256.o ../../crypto/micro-ecc/uECC.o ../../crypto/tiny-AES-c/aes.o -mcpu=cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -mcpu=cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -mcpu=cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -L../../tinycbor/lib -specs=nano.specs  -specs=nosys.specs  -Wl,--gc-sections -u _printf_float -lnosys -Tstm32l432xx.ld -Wl,-Map=solo.map,--cref -ltinycbor -o solo.elf
../../tinycbor/lib/libtinycbor.so: file not recognized: File format not recognized
collect2: error: ld returned 1 exit status
Makefile:65: recipe for target 'solo.elf' failed
make: *** [solo.elf] Error 1
```
I run command:
```bash
$ file ../../tinycbor/lib/libtinycbor.so
../../tinycbor/lib/libtinycbor.so: symbolic link to libtinycbor.so.0.5.2
$ file ../../tinycbor/lib/libtinycbor.so.0.5.2
../../tinycbor/lib/libtinycbor.so.0.5.2: ELF 64-bit LSB shared object, x86-64, version 1 (SYSV), dynamically linked, BuildID[sha1]=f5aa03f58f991617c6660428203c593d1b3883d7, not strippedv
```
It mean libtinycbor.so.0.5.2 is builded for 64bit machine, not for STM32L442
Please guide me how to compiler this firmware.
Thanks

nickray, 2018-11-29:
I think you have to `make clean` at the parent level, and then `make` inside the stm32 target. It could definitely be improved :)

tiena2cva, 2018-11-30:
> I think you have to `make clean` at the parent level, and then `make` inside the stm32 target. It could definitely be improved :)

I have done this but I still got same error. I delete solo folder and re-clone project
```bash
git clone --recurse-submodules https://github.com/SoloKeysSec/solo
cd solo/
git submodule init
git submodule update

cd targets/stm32l442
make all
```
but I got error
```
/home/t-u/Tools/gcc-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld: cannot find -ltinycbor
collect2: error: ld returned 1 exit status
Makefile:65: recipe for target 'solo.elf' failed
make: *** [solo.elf] Error 1
```
because tinycbor is not builded
I build tinycbor with `arm-none-eabi-gcc` as I said above
```bash
cd ,,/,,/tinycbor
make CC=arm-none-eabi-gcc LDFLAGS="-specs=nosys.specs"
```
and rebuild `targets/stm32l442`
```bash
cd ../targets/stm32l442
make clean
make all
```
but still error
```
arm-none-eabi-gcc src/main.o src/init.o src/redirect.o src/flash.o src/rng.o src/led.o src/device.o src/fifo.o src/crypto.o src/attestation.o src/startup_stm32l432xx.o src/system_stm32l4xx.o lib/stm32l4xx_ll_rng.o lib/stm32l4xx_ll_tim.o lib/stm32l4xx_ll_gpio.o lib/stm32l4xx_hal_pcd.o lib/stm32l4xx_ll_usb.o lib/stm32l4xx_hal_pcd_ex.o lib/stm32l4xx_ll_pwr.o lib/stm32l4xx_ll_rcc.o lib/stm32l4xx_ll_usart.o lib/stm32l4xx_ll_utils.o lib/usbd/usbd_hid.o lib/usbd/usbd_conf.o lib/usbd/usbd_core.o lib/usbd/usbd_ctlreq.o lib/usbd/usbd_ioreq.o lib/usbd/usbd_desc.o ../../fido2/util.o ../../fido2/u2f.o ../../fido2/test_power.o ../../fido2/stubs.o ../../fido2/log.o ../../fido2/ctaphid.o ../../fido2/ctap.o ../../fido2/ctap_parse.o ../../fido2/main.o ../../crypto/sha256/sha256.o ../../crypto/micro-ecc/uECC.o ../../crypto/tiny-AES-c/aes.o -mcpu=cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -mcpu=cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -mcpu=cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -L../../tinycbor/lib -specs=nano.specs  -specs=nosys.specs  -Wl,--gc-sections -u _printf_float -lnosys -Tstm32l432xx.ld -Wl,-Map=solo.map,--cref -ltinycbor -o solo.elf
/home/t-u/Tools/gcc-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld: error: required section '.rel.plt' not found in the linker script
/home/t-u/Tools/gcc-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld: final link failed: Invalid operation
collect2: error: ld returned 1 exit status
Makefile:65: recipe for target 'solo.elf' failed
make: *** [solo.elf] Error 1
```

conorpp, 2018-11-30:
From `targets/stm32l442`, run

```
make cbor   # build cbor library
make           # build application
```

TODO:
* [ ] document this better

tiena2cva, 2018-11-30:
@conorpp Thank for your reply.
T follow your command but I still got error: `error: required section '.rel.plt' not found in the linker script`
I search this error on google and see many people got it (most expected in embedded projects, cross-compile) but I don't see any answer.
Do I miss something?

tiena2cva, 2018-11-30:
oh, I got a little error: 
```
../../crypto/sha256/sha256.c:17:20: fatal error: memory.h: No such file or directory
 #include <memory.h>
                    ^
compilation terminated.
```
but I fix it by commend `#include <memory.h>` line in `crypto/sha256/sha256.c` file.
Does it have any effect to above error?

tiena2cva, 2018-11-30:
About my tool chain:
I download [tool chain](https://launchpad.net/gcc-arm-embedded/+download) and use this
```bash
$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 5.4.1 20160919 (release) [ARM/embedded-5-branch revision 240496]
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```
Have any mistake there?

closed 2018-11-30 by tiena2cva