solokeys/solo1 #40: Kickstarter is over

reedy, 2018-12-07:
https://github.com/SoloKeysSec/solo#how-do-i-get-one is therefore out of date meaning https://solokeys.com/preorder/ is the only way to buy one now

Docs need an update :)

conorpp, 2018-12-14:
Thanks, we will update this soon :)  Currently pretty swamped with fulfilling KS orders

reedy, 2018-12-14:
Good to hear (you're busy)!

Slightly offtopic... Any idea what the post kickstart preorder shipping delays look like? Obviously as the page says, the solo tap are coming later... But I'm wondering about ordering a few of the various types to be delivered to an address in the US for me to pick up in late Jan/early Feb and save me some tax/vat/import hassle :)

stappersg, 2019-10-17:
The https://github.com/SoloKeysSec/solo#how-do-i-get-one (note the **Sec**) redirects to https://github.com/solokeys/solo#how-do-i-get-one which is just https://github.com/solokeys/solo

Please close this issue.  Asking @reedy for help  (because issue reporters can close the issues they reported)

closed 2019-11-03 by 0x0ece