solokeys/solo1 #41: opensource hardware

char32, 2018-12-10:
Hey,
is there a plan to opensource the hardware too like with the U2FZero? The main page states "easy to solder hardware" but I mean does it include gerbers, KiKad, etc.? I enjoyed that with the U2Fzero.
In the meantime I'll start to get a nucleo board :-)

Thanks!

- char32

conorpp, 2018-12-14:
Yup, I plan to open source the hardware soon, at least before Kickstarter orders are delivered.  First revision will be same as what we manufactured.  Uses QFN32 package and a fine pitch 4 pin part, so not quite as easy to solder as U2FZero for now, but it's pretty doable with a stencil and hot air.

craftyguy, 2018-12-21:
Any update on this?

conorpp, 2018-12-30:
Shipping and holidays delayed me, but I just open sourced it here:
https://github.com/SoloKeysSec/solo-hw

Sorry for delay!

closed 2018-12-30 by conorpp