solokeys/solo1 #44: MacOS Compilation Error for cbor

wakefieldite, 2018-12-20:
Following [Building](https://solo.solokeys.io/building/)
make cbor:
```
make -f application.mk -j8 cbor
make[1]: Entering directory '/Users/$foo/Documents/GitHub/solo/targets/stm32l442'
cd ../../tinycbor/ && make clean
make[2]: Entering directory '/Users/$foo/Documents/GitHub/solo/tinycbor'
make[2]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
Makefile:95: warning: funopen and fopencookie unavailable, open_memstream can not be implemented and conversion to JSON will not work properly!
rm -f src/cborerrorstrings.o src/cborencoder.o src/cborencoder_close_container_checked.o src/cborparser.o src/cborpretty.o src/cborparser_dup_string.o src/cborpretty_stdio.o src/cbortojson.o src/cborvalidation.o
rm -f src/cborerrorstrings.pic.o src/cborencoder.pic.o src/cborencoder_close_container_checked.pic.o src/cborparser.pic.o src/cborpretty.pic.o src/cborparser_dup_string.pic.o src/cborpretty_stdio.pic.o src/cbortojson.pic.o src/cborvalidation.pic.o
rm -f tools/cbordump/cbordump.o
rm -f bin/cbordump
rm -f bin/json2cbor
rm -f lib/libtinycbor.a
rm -f lib/libtinycbor-freestanding.a
rm -f tinycbor.pc
rm -f lib/libtinycbor.so*
test -e tests/Makefile && make -C tests clean || :
make[2]: Leaving directory '/Users/$foo/Documents/GitHub/solo/tinycbor'
cd ../../tinycbor/ && make CC="arm-none-eabi-gcc" AR=arm-none-eabi-ar \
LDFLAGS="-mcpu=cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -L../../tinycbor/lib -specs=nano.specs  -specs=nosys.specs  -Wl,--gc-sections -u _printf_float -lnosys" \
CFLAGS="-Isrc/ -Isrc/cmsis/ -Ilib/ -Ilib/usbd/ -I../../fido2/ -I../../fido2/extensions -I../../tinycbor/src -I../../crypto/sha256 -I../../crypto/micro-ecc -I../../crypto/tiny-AES-c -c -DDEBUG_LEVEL=0 -DSTM32L442xx -DAES256=1  -DUSE_FULL_LL_DRIVER -DAPP_CONFIG=\"app.h\"    -Wall -fdata-sections -ffunction-sections -mcpu=cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -g"
make[2]: Entering directory '/Users/$foo/Documents/GitHub/solo/tinycbor'
make[2]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
Makefile:95: warning: funopen and fopencookie unavailable, open_memstream can not be implemented and conversion to JSON will not work properly!
arm-none-eabi-gcc -D_FORTIFY_SOURCE=2 -mmacosx-version-min=10.9 -I./src -std=gnu99 -Isrc/ -Isrc/cmsis/ -Ilib/ -Ilib/usbd/ -I../../fido2/ -I../../fido2/extensions -I../../tinycbor/src -I../../crypto/sha256 -I../../crypto/micro-ecc -I../../crypto/tiny-AES-c -c -DDEBUG_LEVEL=0 -DSTM32L442xx -DAES256=1  -DUSE_FULL_LL_DRIVER -DAPP_CONFIG="app.h"    -Wall -fdata-sections -ffunction-sections -mcpu=cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -g  -c -o src/cborerrorstrings.o src/cborerrorstrings.c
**arm-none-eabi-gcc: error: unrecognized command line option '-mmacosx-version-min=10.9'**
make[2]: *** [Makefile:223: src/cborerrorstrings.o] Error 1
make[2]: Leaving directory '/Users/$foo/Documents/GitHub/solo/tinycbor'
make[1]: *** [application.mk:82: cbor] Error 2
make[1]: Leaving directory '/Users/$foo/Documents/GitHub/solo/targets/stm32l442'
make: *** [Makefile:58: cbor] Error 2
```
arm-none-eabi-gcc --version:
```
arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2018-q2-update) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

Actual MacOS version: 10.14.2

0x0ece, 2018-12-20:
I'm still on High Sierra unfortunately so I can't repro in the short time.

Are you able to remove that option `-mmacosx-version-min=10.9` from tinycbor build process?
As an alternative you could try building on a linux Docker container?

wakefieldite, 2018-12-20:
Unset CPPFLAGS for build, seemed to work
`$ unset CPPFLAGS`
go through compile process, merged hex files, and re-set CPPFLAGS environment variable.
`$ export CPPFLAGS='-D_FORTIFY_SOURCE=2 -mmacosx-version-min=10.9'`

I'll keep this open in case there are any other issues that come up in the process of testing.
If all goes well, I will close this issue. Thanks for the advice @0x0ece

0x0ece, 2018-12-20:
Yes, good idea, let's keep it open until we have a smooth make process on Mac.

wakefieldite, 2018-12-21:
For the sake of Googling, here are some of the errors I had going forward.
`STM32CubeProgrammer.app/Contents/MacOs/bin/STM32_Programmer_CLI -c port=USB1 -halt -d /Users/$foo/Documents/GitHub/solo/targets/stm32l442/all.hex`
 **USB1 needs to be lowercase, but that process is still not ready to be done.**

```
-------------------------------------------------------------------
                        STM32CubeProgrammer v1.3.0
      -------------------------------------------------------------------

Error: Cannot open port USB, it may be used by another application or port name is invalid.
```
From [Building](https://solo.solokeys.io/building/)
```
# Makes it boot to ST DFU once
python tools/programmer.py --st-dfu
```
Errors that come up need fido2 and ecdsa installed so:
`pip install fido2 ecdsa`

**Now you can run the programmer.py above to get into dfu mode**
once set in DFU mode, you will notice the light on your Solo Key is no longer on.

Now, I probably shouldn't have run:
`STM32CubeProgrammer.app/Contents/MacOs/bin/STM32_Programmer_CLI -c port=usb1 -halt -d /Users/$foo/Documents/GitHub/solo/targets/stm32l442/all.hex`

Because 
>To program a Solo Hacker device, run the following. Note you should only specify the application firmware, not the combined bootloader+application! I.e. not all.hex from above.
`python tools/programmer.py target/stm32l442/solo.hex`

Despite the return from STM32_Programmer_CLI:
```
      -------------------------------------------------------------------
                        STM32CubeProgrammer v1.3.0
      -------------------------------------------------------------------

USB speed   : Full Speed (12MBit/s)
Manuf. ID   : STMicroelectronics
Product ID  : STM32  BOOTLOADER
SN          : 205F336C4648
FW version  : 0x011a
Device ID   : 0x0435
Device name : STM32L43xxx/STM32L44xxx
Device type : MCU
Device CPU  : Cortex-M4

Warning: halt command is not supported with this port: usb1

Memory Programming ...
Opening and parsing file: /
  File          : /
  Size          : 82824 Bytes
  Address       : 0x08000000

Erasing memory corresponding to segment 0:
Erasing internal memory sectors [0 7]
Erasing memory corresponding to segment 1:
Erasing internal memory sectors [10 43]
Erasing memory corresponding to segment 2:
Erasing internal memory sector 108
Erasing memory corresponding to segment 3:
Erasing internal memory sector 113
Download in Progress:
[==================================================] 100%

File download complete
Time elapsed during download operation: 00:00:02.057
```

Because now, when trying to write

0x0ece, 2019-01-15:
@wakefieldite did you fix? I just upgraded to Mojave, all working fine for me.
I'd recommend a make clean, and then re-follow the instructions in the readme step by step. Also other people have had issues w/ python2, current instructions use python3.

(Also, please remember to hold down the button for a couple seconds when you plug Solo Hacker in.)

wakefieldite, 2019-12-11:
@0x0ece I was able to get the solokey A's to work via building on Arch Linux, however it seems like there is something wrong with the USB-C versions. I'll make a different issue for that so we can close this issue and keep things on topic.

closed 2019-12-11 by wakefieldite