solokeys/solo1 #50: Linux builds don't work

nickray, 2019-01-05:
Building current master on Linux produces HEX files that "don't work".

As a workaround, I created a [release](https://github.com/SoloKeysSec/solo/releases/tag/basic-hacker-build) based on a Windows build by Conor.

We should obviously fix this ASAP.

conorpp, 2019-01-06:
was trying to link cbor dynamically, which obviously crashes the firmware when it runs.  fixed in 83d59df

closed 2019-01-06 by conorpp

nickray, 2019-01-06:
Can confirm! :tada:

The other potential issue: Do not use debian's `arm-none-eabi-gcc` and `libnewlib-arm-none-eabi`packages, download [from arm](https://developer.arm.com/-/media/Files/downloads/gnu-rm/8-2018q4/gcc-arm-none-eabi-8-2018-q4-major-linux.tar.bz2?revision=d830f9dd-cd4f-406d-8672-cca9210dd220?product=GNU%20Arm%20Embedded%20Toolchain,64-bit,,Linux,8-2018-q4-major) directly, and either add its `bin/` to PATH or use `make (...) PREFIX=/path/to/arm/gnu/bin/` to build.