solokeys/solo1 #55: Use separate VID/PID for Solo Bootloader and Solo Hacker

nickray, 2019-01-07:
We have been assigned two VID/PID pairs:
- https://github.com/pidcodes/pidcodes.github.com/pull/399
- http://pid.codes/org/SoloKeys/

I suggest we use 0x1209/0x50B0 for the bootloader, and 0x1209/0x5070 for the hacker build, to distinguish from 0x0483/0xA2CA which we'd continue to use for the locked down regular consumer Solo.

It seems to be a bit more involved than just adding some more defines to `lib/usbd/usbd_desc.c` as I managed to brick a key that way :)

yparitcher, 2019-01-07:
works see #51

conorpp, 2019-01-07:
why not just use a different serial number or other string descriptor?  I think using 3 different VID/PIDs will add more complication.  E.g. now the Linux udev rule needs to be updated

yparitcher, 2019-01-07:
> why not just use a different serial number or other string descriptor? I think using 3 different VID/PIDs will add more complication. E.g. now the Linux udev rule needs to be updated

i agree, now that i realize how it affects udev. i can revert this in my pull request.

nickray, 2019-01-08:
- i think we should reserve the serial number for use as an actual serial number (e.g. if we expose STM32_UUID optionally for some customers)
- i'm unhappy with using "product" or other descriptive/cosmetic strings in the rules, it's kind of brittle (we've already changed it in the past and may continue to do so e.g. to be more systematic).
- the udev rule actually becomes simpler if we can use vid/pid as identifiers directly to spec access to the bootloader (1209/50B0), solo (0483/A2CA), or experimental (1209/5070) hacker or other build.
- i think the official udev rule for consumers should only enable the "real" Solo (maybe setting NAME instead of just adding a SYMLINK)
- for hacker, you could disable the bootloader rule to make it harder for malware to infect your hacker key
- using pid 0483 becomes invalid if we switch chip (manufacturer)
- solo bootloader has a lot of potential for use/love/contributions outside of solo proper. from my bricked key experience i have a hunch that something (either in solotool, bootloader or application) relies on bootloader and application having the same vid/pid, which is not nice to someone wanting to build their separate application with their own vid/pid pair
 
obviously it complicates the code (generation) a bit. the advantage should be conceptual and reading clarity gains if we pull config parameters out of a C file in a nested subdirectory. i have a big thing for disentangling/separation of concerns...

all in all it's a bit of an opinion/taste thing i guess :-)

conorpp, 2019-01-08:
I think it's more simple to use the same vid/pid for all of them.  Even if malware can infect the key, you shouldn't be using a "hacker" key for security purposes.  Not sure of an other use to distinguish in a udev file.

We could also distinguish between builds using the version parameter in the U2FHID protocol, or adding a custom command.  Easier to handle from a scripting or Browser-side perspective.

nickray, 2019-02-03:
Main goal is to easily distinguish hacker vs bootloader (e.g. via `dmesg`). Otherwise it can be very confusing. Maybe abusing serial number is the easiest way.

conorpp, 2019-02-12:
On ubuntu, when I plug in a Solo, I see this

```
[   23.408248] usb 3-1: USB disconnect, device number 2
[   25.951516] usb 3-1: new full-speed USB device number 4 using xhci_hcd
[   26.080583] usb 3-1: New USB device found, idVendor=0483, idProduct=a2ca
[   26.080590] usb 3-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   26.080594] usb 3-1: Product: Solo HACKER (Unlocked)
[   26.080597] usb 3-1: Manufacturer: Solo Keys
[   26.080600] usb 3-1: SerialNumber: 0123456789ABCDEF
[   26.080789] usb 3-1: ep 0x81 - rounding interval to 32 microframes, ep desc says 40 microframes
[   26.080794] usb 3-1: ep 0x1 - rounding interval to 32 microframes, ep desc says 40 microframes
[   26.081949] hid-generic 0003:0483:A2CA.0004: hiddev0,hidraw0: USB HID v1.11 Device [Solo Keys Solo HACKER (Unlocked)] on usb-0000:00:14.0-1/input0
```

Maybe just change the product string?  Can also put a version string in there.

nickray, 2019-02-13:
We have the version in the product string now.

closed 2019-02-13 by nickray