solokeys/solo1 #57: Feature Request: Porting to nRF52840

SandeepanSengupta, 2019-01-07:
Hi @nickray , Is there any way to port it to **nRF52840** as it have all the needed features (Bluetooth, NFC and USB).
It would be great if you guys have a look, specially **[this](https://store.makerdiary.com/collections/frontpage/products/nrf52840-mdk-usb-dongle)** one as it have  a convenient form factor and readily accessible to other makers

nickray, 2019-01-08:
I am the wrong person to ask for C ports :)

caizelin, 2019-01-08:
@SandeepanSengupta The U2F works on nRF52840 now. See the [nrf52-u2f](https://github.com/makerdiary/nrf52-u2f) repo. And I think it's possible to port solo to the [nrf52840-mdk-usb-dongle](https://github.com/makerdiary/nrf52840-mdk-usb-dongle).

0x0ece, 2019-01-09:
There’s a partial port of Solo for the Nordic already, see code inside
targets/. I’m not sure the state, but originally we were thinking to this
chip for Solo, then considered SAM L11, and finally landed to the STM L4.

To my understanding, all MCU but the STM required an external chip to
handle usb (I might be wrong though). I’m saying this just because if you
look at our code, you might find the hid missing.

Thank you both for sharing these pointers, it’s a really interesting
platform to explore in the future.
-- 
Emanuele Cesena
emanuele.cesena@gmail.com

Check out my new security key: https://solokeys.com

conorpp, 2019-01-12:
@SandeepanSengupta cryptographic engine isn't really necessary as most things can be reasonably implemented in software with good enough performance.  Software-based is more flexible as well.  The STM32L4 also has multiple endpoints and we currently use them for a composite device (HID + CDC ACM).

NRF52840 is nice since it has Bluetooth, NFC, and USB.  But the NFC is not a passive solution and the chip is overall to power hungry to power off of something like the AS3956. 

I originally ported FIDO2 application to NRF52840 but decided to switch to something else.  You can see the application in `targets/nrf52840/`.  The FIDO2 "API" has changed a bit since then so it probably won't build, and it's missing some features, like saving state to flash.

cypher386, 2019-01-15:
I'm actually working on this port o. The mdk-dongle to add passwordsafe support for all platforms. I'm still working on my development system but I have the hardware and tools. We'll see what happens.

TaaviE, 2019-06-08:
Are there any updates on this or reasons why the support hasn't been maintained? The dongle is really cheap and would make a really nice security key.

0x0ece, 2019-06-08:
As for SoloKeys, we simply deprioritized porting to multiple platforms because we settled on the stm32 for all our products (solo, tap, and soon somu).

Maintaining portability right now would be at the expenses of building more features, and we feel that at least ssh/gpg are much higher priority, so that's what we're focused on.

TBH I don't see us making a BLE authenticator (with battery) anytime soon, for a variety of reasons including security of BLE, logistic nightmare of batteries, and user support with moving pieces.

This said, porting to nRF52840-MDK and possibly even implementing BLE is something certainly super interesting, if anyone is willing to contribute we're happy to accept.