solokeys/solo1 #67: Signed firmware and GPLv3

jans23, 2019-01-11:
As far as I understand, Solokey is licensed as GPLv3. At the same time (non-hacker) Solokeys can only be updated with signed firmwares but not with user-modified firmware. This restriction violates users' freedom which GPLv3 aims to guarantee. See [Tivoization](Tivoization).

conorpp, 2019-01-12:
SoloKeys is the copyright owner of Solo and doesn't necessarily need to release binary updates as GPL.

The sole motivation for locking down keys is security.  If unsigned updates were allowed, then it would be trivial to duplicate any key with a moment of physical access.  We could make an unsigned update process that erases all sensitive material, but it can still be "backdoored" just as easily.

Unlike Tivo, we want people to be free to make modifications, so they can [either build their own](https://github.com/solokeys/solo-hw) or purchase a non-locked Solo which we distribute as well.

jans23, 2019-01-14:
How I understand your answer is, you basically dual-license your firmware and the firmware you ship on locked devices is not GPLv3 licensed but proprietary. This way you avoid the violation of GPLv3. Of course, technically both firmware are identical. Is this correct?

nightmared, 2019-01-15:
The firmware source isn't proprietary, but it's implementation in hardware HAVE to be signed by a cryptographic key to make Solokeys easy to use. For security purposes, that key obviously can't be made public. If you want to trade ease of use with customization and full control, then, by all means, build your own or buy a "solo hacker", as suggested earlier by @conorpp. I think however you can understand the key to democratization of hardware security tokens IS ease of use.
So trying to enforce strongly the GPLv3 license here is only going to weaken this initiative, and it's already rare enough to see open-source hardware projects like this in the wild. Besides, I'm sure that Solo (the organization/company/whatever behind the project) took take of such issues, as @conorpp explained.

jans23, 2019-01-16:
I'm aware of this challenge and we face the same at Nitrokey. This is one of the reasons why we didn't provide signed firmware updates yet.

I totally agree that the intent is right and that the reasoning "makes sense". But this is not sufficient from a legal point of view. From my understanding my previous [summary](https://github.com/solokeys/solo/issues/67#issuecomment-453933118) is a sufficient solution.

closed 2019-01-18 by jans23