solokeys/solo1 #68: Build error on MacOS at the last step of hacker build

raj-bhatia, 2019-01-11:
I am new to Solo! I am trying to build the firmware for the first time following the instructions on Solo's "Build instructions" page. After some effort (installing several Python packages), I am close to the build process. The "make cbor" command works. The "make build-hacker" command fails near the end. Following is the error I am encountering:

arm-none-eabi-objcopy -O ihex solo.elf solo.hex
python ../../tools/solotool.py mergehex solo.hex bootloader.hex all.hex
('merging solo.hex with ', 'bootloader.hex')
Traceback (most recent call last):
  File "../../tools/solotool.py", line 996, in <module>
    main_mergehex()
  File "../../tools/solotool.py", line 952, in main_mergehex
    first.tofile(args[len(args) - 1], format='hex')
  File "/Library/Python/2.7/site-packages/intelhex/__init__.py", line 719, in tofile
    self.write_hex_file(fobj)
  File "/Library/Python/2.7/site-packages/intelhex/__init__.py", line 685, in write_hex_file
    bin[4+i] = self._buf[cur_addr+i]
TypeError: an integer is required
make: *** [build-hacker] Error 1

I will appreciate any help that can be provided to resolve this issue.

cjsieh, 2019-01-11:
I had a similar problem on Linux.  The instructions say to run
"pip3 install -r tools/requirements.txt" to add the requirements for python.  This implies that python-3 is needed.

On both my system and yours "python" is python 2.7 .  I changed the Makefile to run python 3 . It worked fine after that.

0x0ece, 2019-01-15:
I can confirm it works with the instructions in the readme, thus python3, on high sierra and mojave. I'm closing this now, please reopen if there are more issues.

raj-bhatia, 2019-01-15:
I have installed python3 on the Mac. Python2 is still installed but I have made python3 the default (so when python command is called, it invokes python3). The original issue is resolved but now I am facing another issue. "make build-hacker" now gives the following error:

arm-none-eabi-objcopy -O ihex solo.elf solo.hex
python ../../tools/solotool.py mergehex solo.hex bootloader.hex all.hex
Traceback (most recent call last):
  File "../../tools/solotool.py", line 34, in <module>
    from fido2.hid import CtapHidDevice, CTAPHID
ModuleNotFoundError: No module named 'fido2'
make: *** [build-hacker] Error 1

I have installed fido2, but it seems to be associated with python2 (not python3). Any other guidance will be much appreciated.

0x0ece, 2019-01-15:
Mmm... can you try following the steps in the readme and see if that works?
-- 
Emanuele Cesena
emanuele.cesena@gmail.com

Check out my new security key: https://solokeys.com

0x0ece, 2019-01-15:
Yes that readme. Python is right after (make env3 etc.)

> --
Emanuele Cesena
emanuele.cesena@gmail.com

Check out my new security key: https://solokeys.com

raj-bhatia, 2019-01-15:
Thanks for your help! After I executed the following two commands,

make env3
source env3/bin/activate

my build errors went away. Looks like these commands are needed before attempting compilation. I will close this issue now.

closed 2019-01-15 by raj-bhatia