solokeys/solo1 #72: Option to disable breathing LED Mode

tscs37, 2019-01-17:
At the moment, the SoloKey will enter a "Breathing" mode with it's green LED when it's plugged in an powered, ie, the LED dims off and on continuously.

While in most scenarios this is fine, I would like it if I could somehow disable this behaviour and have the LED be in a static mode as this animation can be a bit distracting if the environment is dark.

conorpp, 2019-01-17:
Right now LED setting is set at compilation.  Could move this to a flash setting and make it updateable from `solotool.py`.  A quick fix is to edit [`targets/stm32l432/src/app.h`](https://github.com/solokeys/solo/blob/master/targets/stm32l432/src/app.h#L56) and set `LED_MAX_SCALER` and `LED_MIN_SCALER` equal to each other, making the "breathing range" 0.

```c
//                              0xRRGGBB
#define LED_INIT_VALUE			0x000800
#define LED_WINK_VALUE			0x000010
#define LED_MAX_SCALER          30
#define LED_MIN_SCALER          1
// # of ms between each change in LED
#define HEARTBEAT_PERIOD        100
// Each LED channel will be multiplied by a integer between LED_MAX_SCALER
// and LED_MIN_SCALER to cause the slow pulse.  E.g.
// #define LED_INIT_VALUE			0x301000
// #define LED_MAX_SCALER          30
// #define LED_MIN_SCALER          1
// #define HEARTBEAT_PERIOD        8
// Will pulse from 0x301000 to 0x903000 to 0x301000 ...
// Which will take ~8 * (30)*2 ms
```

Should the default green breathing be changed?

tscs37, 2019-01-18:
The default is probably fine, a simple flash setting to disable it should be sufficient for people who don't like it IMO.

nickray, 2019-01-18:
I agree that the LED is very bright (annoying even in low-light settings). Such configuration should also be included in the regular Solos, although we need to make sure we don't introduce unsafe APIs :)

tscs37, 2019-01-19:
I don't think the LED flashing is terribly security relevant, the worst I can imagine there is that an attacker could trick someone into believing the device is defective. (I'm not sure if that attack scenario is relevant for solokey, probably not)

It might be useful to require pressing the button on the device to confirm any new flash settings like this.

Square252, 2019-05-08:
Would also be nice if we could change the color of the breathing at flash time.

My1, 2019-08-11:
@Square252 nice idea, so one can have the option to choose a color for example depending on his sleeve.