solokeys/solo1 #73: bogus name for udev rule

bleuse, 2019-01-17:
The provided udev rule sets the `uaccess` TAG to allow the logged user to use the key. This is however useless as the name of the rule is `99-solo.rules`.

According to udev man pages:
> All rules files are collectively sorted and processed in lexical order, regardless of the directories in which they live.

The udev rule from systemd setting ACL based on the `uaccess` tag is `73-seat-late.rules`. Hence, the name of the provided rule should begin with a 2 digit number at most 73. The rule provided by @craftyguy in #45 was originally correct (content and name of the rule).
Looking at the `libu2f-udev` package in debian, setting the tag should be enough when using a recent version of udev. Please also note that plugdev group is deprecated.

I however believe that it would be better if we keep in the documentation a rule setting mode and group for user of older udev versions. This should be properly flagged.

yparitcher, 2019-01-17:
+1
i had the same experience and had to put the rule with an earlier number for it to work

bleuse, 2019-01-18:
@conorpp I'd gladly help on this issue. I have a working rule here. The one provided in the example seems to be over-complex from what I understand of udev.
What do you think of providing the simplest rule(s) for basic users in a rule file, along with a second set of rules in another file for more advances users/uses?
Based on your feedback, I'll propose a PR.

bleuse, 2019-01-18:
> > The one provided in the example seems to be over-complex from what I understand of udev.
> 
> [This one](https://github.com/solokeys/solo/issues/45#issue-393298877) is overly complex?
> 
> Or are you referring to some other example?

I was referring to the example file in the repository as of now (1866d36). IMHO it'd be better to provide a minimal rule (i.e., exactly the one you highlight) without the symlinks, groups, modes, … and a second set of rules for advanced uses.
As all udev rules will be triggered, it has the same effect without adding clutter for basic users.

nickray, 2019-02-26:
@bleuse thoughts on https://github.com/solokeys/solo/pull/119 ?

bleuse, 2019-03-03:
@nickray the merged PR looks good to me, and addresses this issue. I guess we can close the issue.

closed 2019-03-21 by nickray