solokeys/solo1 #74: Not working with Google

Espionage724, 2019-01-17:
https://solokeys.com/en/start/ mentions you can use Solo with Google. However, the 2FA security key page doesn't seem to work.

A Webauthn test page and GitHub work with Solo no issue though, so at the very least, it seems I have the browser support good to go.

Using Firefox 64.0.2 on Fedora Workstation 29. I haven't tested with Chrome, but will give it a try at some point and update how that works.

ignaloidas, 2019-01-17:
Google just doesn't work with Firefox using any key, as they use some Chrome specific thing when authenticating. Same with Facebook.

0x0ece, 2019-01-17:
I believe Google only supports registration on Chrome. Then you can use it
on Firefox too.

-- 
Emanuele Cesena
emanuele.cesena@gmail.com

Check out my new security key: https://solokeys.com

Espionage724, 2019-01-17:
> I believe Google only supports registration on Chrome. Then you can use it on Firefox too.

Yup that's what was needed. Never got the U2F prompt when trying to register the key on Firefox, but got it no problem on another Fedora Workstation 29 machine with Google Chrome (no udev rule or config needed). Then from there, the key works fine for logging in on Firefox.

closed 2019-01-17 by Espionage724