solokeys/solo1 #76: Outdated Google info on "start" webpage

ahwayakchih, 2019-01-18:
On https://solokeys.com/en/start/ page there is:

```
1. Visit https://myaccount.google.com and click on Signing in to Google.
2. Click on 2-Step Verification to start the registration process.
```

I'm not sure if it's just me, or if Google updated it everywhere, but i see a menu on the left side of the "my account" page, and various blocks on right/main side. None of them is "Signing in to Google".
Only after i click on "Security" menu item, i can see "Signing in to Google" section/block (2nd one, after "We keep your account protected" section), which contains "2-Step Verification" link. After clicking it, i see a page with various 2nd steps, one of which is "Security Key" with "Add security key" link.

I know: not much of a difference, especially for, i assume, more "techie" audience of Solo keys at the moment. Still, maybe some "regular" folks will want to try it too :).

nickray, 2019-01-21:
From https://solokeys.com/ you can directly click through to the various websites' own instructions, which should probably be more reliable :)

[image]

ahwayakchih, 2019-01-22:
Indeed. So maybe "start" page should simply encourage to select one of those links, instead of confusing potential users with outdated information?

stappersg, 2019-10-17:
For which reason is this issue still open?

ahwayakchih, 2019-10-17:
Start page changes its URL to https://static.solokeys.com/en/start/, but it still shows up incorrect information.