solokeys/solo1 #77: GPLv3

yparitcher, 2019-01-18:
as already discussed in #67 the GPLv3 does not allow for tiviozation. at the same time Solo needs to be able to distribute signed firmware to most users for security.
therefore Solo is forced to use workarounds in the form of [CLAs](https://solokeys.com/legal/contributors/) to assign rights of contributors back to Solo to use however Solo wants to allow signed firmware thereby circumventing the GPLv3. 

I think a better way implement this would be to licence under the **GPLv2** which is a proper open source licence, but does not explicitly prevent tiviozation which in this case is good, as it allows Solo to distribute signed firmware. This is the path the linux kernel takes, and allows it to be used in secure boot applications.

this has the advantages that contributors are not required to licence their work separately to Solo rather their work remains fully under an FOSS licence, and will will streamline the contribution process.

onlykey, 2019-02-02:
I would recommend using LGPLv3. As yparitcher mentions you are going to have issues using GPLv3 if you are not allowing users to sign their own firmware (Which you definitely don't want to do for security reasons). You will probably also in the future have issues where you really want to use a library that is BSD 4 clause (OpenSSL had issues with this) that is GPL incompatible, so you will have issues using it in a GPLv3 project. 

You can see that other devices which require similar functionality use LGPLv3 - http://blogs.directinsight.co.uk/qt-lgpl-vs-commercial/

"In practice this means that for a typical i.MX6 module-based project, we can use the i.MX6 Secure Boot facility to prevent Device Vulnerability by creating a signed image, with the signature blown into unreadable hardware fuses while complying with LGPLv3."

I would also suggest LGPL because I would like to use some of the Solo Key libs but I also need GPL incompatible libraries. So sticking with GPLv3 means I would have to fork Solo Key at the commit prior to this https://github.com/solokeys/solo/commit/bbc61d5743b45d62f7e2cbd5f0d605a17a890a03

yparitcher, 2019-02-03:
LGPLv3 does not help anything as solo is a (physical) user product not a library.

in order not to have issues with tiviozation the license needs to be v2, either GPLv2 or LGPLv2.

conorpp, 2019-02-12:
We've been discussing a lot internally and are thinking about moving to a more permissive setup.  Probably MIT+Apache2 dual license.  Any thoughts on that?

yparitcher, 2019-02-12:
MIT+Apache2 would be great. I only suggested GPLv2 as i was under the impression that you specifically wanted a GPL type licence.

onlykey, 2019-02-12:
@conorpp @yparitcher Agree, MIT+Apache2 would be great.

conorpp, 2019-02-12:
we're now MIT+Apache2!

closed 2019-02-12 by conorpp