solokeys/solo1 #83: dead cycle

merlokk, 2019-01-21:
looks like here is a dead cycle:
https://github.com/solokeys/solo/blob/master/fido2/log.c#L92
am i right?

conorpp, 2019-01-22:
What do you mean dead cycle?

merlokk, 2019-01-22:
if we "have lost" TAG_ERR we will exec printf2 it calls LOG it comes to this line and repeat)
stack overflow.

secondly maybe not so good make recursion at microcontroller (if TAG_ERR is ok it will call another time LOG()...  )
and i dont know what will did that exit(1) at microcontroller but i dont think it will be good

conorpp, 2019-02-12:
exit(1) triggers an infinite loop on MCU, which is good in that it's deterministic.

I think it won't trigger an infinite loop because `TAG_ERR` is a valid tag, and won't cause the `if (i != 0)` to evaluate an extra time.  It's probably more clean to implement this without `printf2` though, so i'll leave open until it's fixed.

closed 2019-03-19 by conorpp