solokeys/solo1 #84: standalone app

meganchen, 2019-01-21:
I cloned the repo, and I'm following the `Developing Solo (No Hardware Needed)` section of the readme. 

After I run `./main`, the terminal output is 
```state file exists
[STOR] Auth state is initialized
[STOR] pin not set.

```
If I run `python tools/ctap_test.py`, the output shows 
```
Test init
Traceback (most recent call last):
  File "tools/ctap_test.py", line 828, in <module>
    t.test_hid()
  File "tools/ctap_test.py", line 156, in test_hid
    r = self.send_data(CTAPHID.INIT, '\x11\x11\x11\x11\x11\x11\x11\x11')
  File "tools/ctap_test.py", line 85, in send_data
    return self.dev.call(cmd, data, event)
AttributeError: Tester instance has no attribute 'dev'
```

Then, if I run `python python-fido2/examples/credential.py`, the output says 
```No FIDO device found```

Not sure what the desired outputs should be - could you help troubleshoot what's going on? (Not sure if this matters, but I have Mojave). Thanks!

meganchen, 2019-01-23:
I believe I installed the patched version of `python-fido2` (head at `329434f`). 
I checked `tools/ctap_test.py` and noticed I was running `t.test_hid()` instead of `t.find_device()` because I was fiddling around with it. Whoops. Now when I run `python tools/ctap_test.py`, I get 
```
[]
Traceback (most recent call last):
  File "tools/ctap_test.py", line 827, in <module>
    t.find_device()
  File "tools/ctap_test.py", line 73, in find_device
    raise RuntimeError('No FIDO device found')
RuntimeError: No FIDO device found
```
A step that I forgot I did about: I changed line 75 in `pc/device.c` from
```
    serveraddr.sin_port = htons( 8111 );
```
to
```
    serveraddr.sin_port = htons( 0 );
```
to fix an error that said `bind failed: Address already in use`. Also did the same in `tools/http2udb.py` line 44 (`udpport = 8111` -> `udpport = 0`). I wonder if the issue is something to do with which ports `tools/ctap_test.py` is using to find the device.

conorpp, 2019-01-24:
The `bind failed: Address already in use` was due to an error with the routine being called twice.  Just merged this PR which should fix it.

https://github.com/solokeys/solo/pull/58

I would revert `serveraddr.sin_port = htons( 0 );`, it must be 8111.

meganchen, 2019-01-24:
Sweet. Thanks! The `bind failed: Address already in use` error is gone now. I'm still getting `RuntimeError: No FIDO device found`. Not sure what else to try

closed 2019-11-03 by 0x0ece