solokeys/solo1 #90: Supply chain hardening/hardware verification

Vaesper, 2019-01-25:
I just received a padded envelope from Hong Kong Post, containing several ziploc bags. For such a security-crucial device I had expected some kind of anti-tamper packaging. Is there a guide somewhere that would let me at least confirm the PCB I received matches the design by visual means?

0x0ece, 2019-01-26:
@Vaesper as we're looking into packaging, can you make a couple examples of what you'd consider good?

FWIW, all yubikeys I've got just had a seal that you can buy on alibaba, and the secure clicks are sold in a regular paper box.

In terms of security, regular Solo is locked down, the only known way to perform an attack is to decap the STM32 that would be visible. Solo Hacker, viceversa, is unlocked and you have access to the ST bootloader, so you can reflash it entirely. In addition to all this, we're working to document a secure procedure to "factory reset" and regenerate the master secret using the TRNG.

To be honest, I don't think packaging can add a lot of real security, but I totally hear you that the first impression also matters a lot. Cost will definitely be a variable to consider, we don't want to raise prices just because of packaging.

Vaesper, 2019-01-26:
> can you make a couple examples of what you'd consider good?

I'm not an expert in these matters, so please don't take my word as gospel on this, but something I can imagine is sealed packaging with a serial number embedded in it, and the serial number being sent to the buyer in the confirmation email.

> In terms of security, regular Solo is locked down, the only known way to perform an attack is to decap the STM32 that would be visible.

Or replace the device with a fake during shipping. This is why I'm asking for images of what the board should look like so I can do a visual comparison.

conorpp, 2019-01-26:
I just added some images that you can use:

https://github.com/solokeys/solo-hw

A more official way is to verify the attestation certificate (returned in U2F/FIDO2 registration step).  I'll work on adding this to our python tool soon.

conorpp, 2019-01-26:
I just added it.  You can verify firmware is from SoloKeys by doing the following

```python
python tools/solotool.py solo --verify-solo
```

nickray, 2019-01-27:
In my case, this method doesn't exist: https://github.com/solokeys/solo/blob/7a6abdfd0c47233d9545bc2ad7fd27913588b97b/tools/solotool.py#L243

conorpp, 2019-01-27:
I was on 0.4.0 version of `python-fido2` and they changed the method in `0.5.0` (current).  I just updated.

Vaesper, 2019-02-06:
Thanks for the response (and apologies for the slow reply). Closing issue because pictures are now present in the (hw) repo.

To illustrate some of the attack vectors related to this see this twitter thread: https://twitter.com/x0rz/status/1093071488117284864

closed 2019-02-06 by Vaesper