solokeys/solo1 #92: No longer can sign into sites with U2F

Espionage724, 2019-01-30:
Trying to sign into Google and GitHub (only two things I have U2F set-up with); they both don't accept the security key anymore. Was working fine last night.

For GitHub, it says " Security key authentication failed." For Google, it just loops back to the 2FA screen and tells me to press the button again.

conorpp, 2019-01-30:
What browser?  Version?

conorpp, 2019-01-30:
Hmm I'm not able to reproduce on FF 65.  Is security.webauth.u2f set to true in "about:config"?

Espionage724, 2019-01-30:
Yeah I have that option enabled; I also tried from Chrome on the same computer and have the same problem.

conorpp, 2019-01-30:
If you delete the key and re-add it, does the problem persist?  If not, there might be something more subtle going on

Espionage724, 2019-01-30:
No; deleting the previous security key and re-adding for GitHub works fine (auths on both computers from Firefox)

I'd have to go through Google's support to remove the security key to see if re-adding works there (they don't allow fallback to 2FA codes it seems)

conorpp, 2019-01-30:
Hmm this could be a pretty subtle issue.  Can you post a timeline of actions you did with your key?  

E.g. 

day 1 registered to Google, Github
day 2 works auth Google & Github 
day 3 registered to something else?
day 4 auth Google & github stop working.

conorpp, 2019-02-15:
I'm 99% sure I've found and fixed the issue for this.

I had messed up the endianness for the counter for U2F (FIDO2 is fine).  So it would count like this:
```
0x01000000
0x02000000
...
0xff000000
0x00010000
```

So after a number of uses, any service that checks that the counter increases for U2F, will see an issue.  Re-registering fixes it for a max of about ~255 uses.

@nickray and I meanwhile have been working on releasing our first signed update, so any device can be updated from Chrome, Firefox, or our solotool.py.  Will announce when the update is ready.

Updating preserves any previous registrations, but it will mess up anything tracking U2F counters.  So I recommend re-registering them after updating.

Espionage724, 2019-03-12:
At some point a few days ago, I wasn't able to sign into Google again. After a reboot though, it worked fine.

Today I ran into the same problem of not being able to sign into Google, but this time it didn't work after a reboot either. I went to GitHub to see if I could sign-in there, and was able to no problem. I then loaded Google up in another tab and tried signing-in and it worked.

Is there anything locally on my end that might cause U2F sign-in to not work consistently?

conorpp, 2019-03-12:
I'm not sure what could be happening, but I've released a couple new updates since.  Could you try updating to `2.0.0`?

```
solo key update --secure --alpha
```

conorpp, 2019-04-14:
This issue should be fixed with latest updates, please reopen if not.

closed 2019-04-14 by conorpp