solokeys/solo1 #103: USB A key connected to a MBP via a USB C hub times out

benstegink, 2019-02-08:
I have a 2018 MacBook Pro (USB C only). I recently received a USB A solo key. However, when I tried to set it up with Facebook and Dropbox, it fails with a request timeout error. I tried:

- A CalDigit docking station (tried multiple USB A ports on it)
- A USB C -> USB A adapter dongle
- A mini USB C docking station (with multiple USB A port tried)

All of them with the same issue

Wanting to verify if it was the key or not, I tried a Windows 10 computer with USB A ports on it, the key registered with Facebook immediately without any problem

Also tested on my wife's MacBook Air (also with USB A) and was able to register with Google right away without issue.

For all the tests I used the latest version of Google Chrome.  The only time it doesn't seem to work is using a USB C/Thunderbolt 3 hub or some issues specific to my computer. Unfortunately, I don't have access to a second USB C device to test with.

DanielNTX, 2019-02-08:
I tested this on my 2017 MacBook Pro with the Solokeys USB-A (using a Dell USB-C adapter) and USB-C versions and Chrome didn't see the Solokeys.  I just tried with my Yubiko Security Key (FIDO2 version, USB-A) and it worked with no problems.

0x0ece, 2019-02-08:
@benstegink by chance do you have another security key that you could try on your mac? This is to rule out something that may be interfering with a fido hardware token, like a software u2f/fido2, or a chrome extension, or other devices plugged in... One good test could be in Chrome incognito window (i.e. plugin disabled), with just the Solo key plugged in, just to make sure we rule out any possible interference.

@DanielNTX so you have the opposite situation, you have a Solo USB-C but a Mac with a USB-A port, correct? Do you have anything else w/ usb-c to test that the connector carries data, and not just power? I had many similar issues with other adapters, you can charge but not really use them for data.

(thank you both for reporting!)

benstegink, 2019-02-09:
@0x0ece did some more testing this morning
- Chrome incognito mode didn't work
- Firefox worked right away

So, definitely not tied to the USB-C -> USB-A, but actually looks like it might be something related to Chrome.

andrewconnell, 2019-02-10:
@benstegink Try Chrome Canary?

0x0ece, 2019-02-10:
@benstegink Ok, so hw issue is ruled out. For Chrome I assume you have one of the latest versions, right? (I have 71). Also, have you ever tried another security key with your Chrome?

benstegink, 2019-02-11:
@0x0ece, yup, hardware issue ruled out. I'm actually on the beta version (73), I can try to downgrade to 71

@andrewconnell I can also to try upgrade to Canary and see if that works :)

DanielNTX, 2019-02-12:
@0x0ece I have both Solokeys A & C and neither work in macOS 10.14.3 with Chrome 73.0.3683.20 even though I have them registered (works fine in Windows).  I also have a Yubico Feitian FIDO2 key (USB-A) and it works fine to authenticate on GitHub on the mac.

0x0ece, 2019-02-12:
Yubico just sent an email to their developer program with an issue on Chrome 72+.

Workaround 1 - Retry
Try logging in again. This may generate a token without characters that will need to be encoded and authentication can be successful. 

Workaround 2 - Disable Flag for WebAuthenticationProxyCryptotoken
Disabling the WebAuthenticationProxyCryptotoken flag will force Chrome to adopt the U2F behavior prior to the v 72 release. 

To disable the flag, launch Chrome from the command line with the following flag set:
--disable-features=WebAuthenticationProxyCryptotoken

I don't have time to try this out shortly, but if anyone can... (I can also fwd the full email, unfortunately there's no web copy of it to link here -- just drop me a line at ec@solokeys.com)

merlokk, 2019-02-12:
In Chrome v 72 and 73, the issue will cause the RP to receive a response that contains an empty challenge and a signature for the empty challenge. An error similar to the one below will be shown to the user. 

"Your authentication information is incorrect. Please try again."

In Chrome Canary (v 74), the issue will cause a BAD_REQUEST error to be generated. The error message will say: 

   "challenge must be base64url encoded" 
Or:
   "keyHandle must be base64url encoded"

benstegink, 2019-02-12:
Just tried with WebAuthenticationProxyCryptotoken disabled in Chrome v 72 and didn't seem to make any difference for me.

benstegink, 2019-02-15:
Another update, just upgraded to the latest Canary build (v74) and the key is working now with both Facebook and Google. Still doesn't work with Dropbox for some reason, maybe that's actually an issue on the Dropbox site itself?

So, as it stands today just based on tests logging into Google, Facebook, Github, and Dropbox 
- Chrome v72 - seems to work fine (except for Dropbox)
- Chrome v73 - key doesn't work at all
- Chrome v74 - seems to work fine (except for Dropbox)

0x0ece, 2019-02-15:
Thank you for the test, this is great information. We’ll see if we can find someone at Chrome to talk to.

For Dropbox, I’d recommend retrying. I met with the eng who built WebAuthn integration, and we verified it’s working. They apparently have an intermittent bug, but it’s not related to Solo, it fails intermittently with any security key.

fooflington, 2019-02-28:
I am suffering from this, I think. I have USB-A keys being used on a USB-C only MBP. Chrome 72 on macOS (10.14.3) and neither of my keys are working GitHub, Facebook or Google. I've tried browser and full system restarts. The system can see the USB device ok but it just stays on the pulsing green LED.

conorpp, 2019-02-28:
For those having issues, can you update your Solo to the latest version?  Fix a couple bugs with U2F.  Thanks @fooflington for reporting, glad to here the updates solved the problem.

```
pip install -U solo-python
```
```
# update for Solo
solo key update --secure

# update for solo hacker
solo key update --hacker
```

benstegink, 2019-02-28:
@conorpp, thanks! Just updated and the key is working great for me now

xird, 2019-03-05:
I have a very similar issue: A USB-A Solo key, connected to a Dell XPS 13 Ubuntu laptop via a USB-C hub. I'm trying to log in to GitHub with Firefox (Quantum 65.0.1, with the 2FA setting manually set to true), and the authorization times out without the key LED turning yellow.

The USB hub works fine with a keyboard and a trackball.

The key works fine when plugged in directly to a 2013 MacBook Pro.

I've updated the Solo firmware to 1.1.1, but that made no difference.

conorpp, 2019-03-16:
@xird I just noticed you're also testing Ubuntu vs MacOS.  Did you install the udev rules on Ubuntu?  Otherwise only root access works.

https://docs.solokeys.io/solo/udev/

xird, 2019-03-26:
@conorpp That was it. I didn't realize there was configuration needed. Works fine now, thanks!

nickray, 2019-03-26:
For context: we're [working on getting udev rules into Linux distributions](https://github.com/solokeys/solo/issues/144) so this manual setup step can be avoided.