solokeys/solo1 #104: Returning custom data from Solokey

leochou0729, 2019-02-12:
Hello,
  I just want to ask a question here.
  Is it possible to use WebAuthn API to get some custom data from hardware in addition to 256-bit signature? For example, I send in a special RP ID or KEYID. When hardware gets this information, it knows that it's not for authentication and returns another 256-bit information I needs. Then the JS code can use the custom data returned directly and not send it to the server. 
  Does Google Chrome support this kind of usage? Thanks!

merlokk, 2019-02-12:
look here:

https://github.com/solokeys/solo/blob/master/targets/stm32l432/src/app.h#L39

#define ENABLE_U2F_EXTENSIONS

merlokk, 2019-02-12:
dont know how to exec it from browser(  looks like not navigator.credentials.get.

0x0ece, 2019-02-12:
Check out the crypto wallet: https://github.com/solokeys/solo/blob/master/fido2/extensions/wallet.c
and the related js code in `web/`. This should be similar to what you want to do.

I'm not sure about webauthn, we still have to investigate that. If you find out anything, please share :)

conorpp, 2019-02-12:
@nickray can speak more on the webauthn JS API and share some examples, but I believe yes, `navigator.credentials.get()` can be used.  We're planning to document more on how to close the gap between browser and authenticator.

The application you describe is possible and should work on Chrome and Firefox.  We leverage this for our firmware update process and it works well.  Note that the browser will control how RPID is computed based on the domain, so you're limited to KEYID.

nickray, 2019-02-13:
Copying my email answer here:

There's two levels, WebAuthn and CTAP. The latter gives you vendor specific commands [1] in the HID transport, so you can extend communication between client (browser or custom software on your machine) and authenticator.
From the browser, you could use WebUSB to do CTAPHID, but this is only supported Chrome (e.g. Firefox will never implement it, they consider it unsafe).

On the other hand, from the WebAuthn side [2] you only get credential.create and credential.get as you mention.
For our firmware-update-over-web (experimental, WIP), we faced a similar issue. What we do is encode data as a "magic" keyhandle [3] and parse the response.

Actually, in our case we communicate with the bootloader, which only implements U2F/CTAP1, and use the fact that WebAuthn resends CTAP2 commands as CTAP1 if the receiving authenticator does not understand U2F.

[1] https://fidoalliance.org/specs/fido-v2.0-id-20180227/fido-client-to-authenticator-protocol-v2.0-id-20180227.html#usb-vendor-specific-commands
[2] https://webauthn.guide
[3] https://github.com/solokeys/solo-station/blob/27d2b383303e6cdb61424e45d3359da8b2ffdfff/js/ctap1.js#L15

leochou0729, 2019-02-14:
@conorpp @nickray Thanks a lot!

dschuermann, 2019-02-28:
@nickray Very helpful thread. Thanks for the pointers. AFAIK implementing CTAPHID over WebUSB in Chrome will not work since they blocked this device class due to https://www.yubico.com/support/security-advisories/ysa-2018-02/ . Or am I missing something?

merlokk, 2019-02-28:
https://github.com/chromium/chromium/blob/master/chrome/browser/usb/usb_blocklist.cc#L60

u2f zero here)

dschuermann, 2019-02-28:
@merlokk Every U2F/FIDO2 security key should simply be blocked because the HID and CCID USB classes are blocked in https://github.com/chromium/chromium/blob/0aee4434a4dba42a42abaea9bfbc0cd196a63bc1/third_party/blink/renderer/modules/webusb/usb_device.cc#L520 . Or am I missing something? Was the block list based on PIDs even necessary?

dschuermann, 2019-02-28:
>  Was the block list based on PIDs even necessary?

Okay, answering my own question: https://bugs.chromium.org/p/chromium/issues/detail?id=818592#c12