solokeys/solo1 #112: Virtual authenticator does not work (or I'm doing something wrong)

badregged, 2019-02-15:
I am unable to use virtual authenticator, websites do not see it.
For example, on https://www.webauthn.org/ I'm waiting but nothing happens. The same about GitHub.
I am using ./main and Tor Browser Bundle (Firefox‐based). Am I doing anything wrong? Is virtual authenticator broken?
#31 related.

0x0ece, 2019-02-15:
The virtual authenticator is just a convenient way to develop Solo firmware without the need for hardware. It can't be used with browsers or applications.

The way it works is that we patched python-fido2 library to emulate the protocol over UDP. So our virtual authenticator can only be used via this patched python-fido2 library.

badregged, 2019-02-15:
Is there some virtual authenticator for use with browsers?

0x0ece, 2019-02-15:
The only one I know of are:
https://krypt.co
https://github.com/github/SoftU2F

If you find others, please share them, we're always interested in other projects in the space.

badregged, 2019-02-16:
Anything which works on GNU/Linux?

badregged, 2019-02-16:
Found some software: https://github.com/concise/v2f.py
Is it possible to force it to work with passwordless/WebAuthn?

0x0ece, 2019-02-16:
U2F unlikely, you’d need UAF.

Or better FIDO2 that is basically the unification of the two.
-- 
Emanuele Cesena
emanuele.cesena@gmail.com

Check out my new security key: https://solokeys.com

badregged, 2019-02-16:
Looks like UAF is **much less secure** than password:
[image]

1. Biometry is already broken. Fingerprints can be found everywhere.
2. USB with button can be stolen or be used to open the account.
3. USB with password encryption — why pay for USB instead of setting password?

I think,

1. First factor **MUST** be password, always.
2. Second/third factor may be U2F and/or TOTP, which **MUST** be encrypted with a password.

Questions:

1. Why is biometry still trusted, after incidents where fingerprints were found somewhere or fingers were cut?
2. Why are unencrypted MFA devices trusted?

I don't need UAF, but I will not close the issue.

0x0ece, 2019-02-17:
This can be discussed in the FIDO Dev forum, not here.
https://groups.google.com/a/fidoalliance.org/forum/#!forum/fido-dev

closed 2019-02-17 by 0x0ece