solokeys/solo1 #113: Beta Program (for firmware v1.1.0 update)

nickray, 2019-02-18:
## Announcement
We have prereleased our first official update of the Solo firmware: https://github.com/solokeys/solo/releases/tag/1.1.0 :champagne: 

TLDR: `pip install solo-python; solo key update [--secure|--hacker]`

## Invitation
Prior to announcing this update more widely, we invite anyone interested to test the firmware and its update procedure: you can take part in our beta testing program!

Please proceed at your own risk! We will of course provide help in case of any issues.

## Changes
The changes are listed in our https://github.com/solokeys/solo/blob/master/CHANGELOG.md

We will strongly recommend everyone to update, as this fixes a bug in the U2F functionality (FIDO2 functionality was not affected) causing the signature counter to cycle after 255 uses of a given credential.

The firmware release goes hand in hand with the release of `solo-python`, our Python3.6+ library and tool to interact with your Solo. It lives in the repository https://github.com/solokeys/solo-python and can easily be installed via `pip install -U solo-python`. The existing `solotool.py` will be deprecated.

While the official roll-out may use a web-based update procedure for simplicity, by using the Python CLI tool you will be working with a more mature interface.

Additionally, with this release we move to verifiable Docker builds of the firmware, details below.

# How To Update?
- install the Python tool: `pip install -U solo-python`
- update one key at time
- insert the key you wish to update, keeping the button pressed until the LED flashes yellow
- if it is a Solo Secure key ("regular" version): type `solo key update --secure`
- if it is a Solo Hacker key: type `solo key update --hacker`

Please do *not* flash the secure firmware onto a hacker key -- while it will work, this will activate ROP (flash readout protection) level 2, and you will *irreversibly* lose access to the ST DFU interface.

In both cases, the key will automatically be updated. You can verify everything worked by checking the output of the following commands:
```
solo key version  # should show `1.1.0`
solo key verify  # asks you to press the button, checks that the key attests properly
```

## Verifiability
The firmware you will flash by updating is pulled from the v1.1.0 release assets:
- https://github.com/solokeys/solo/releases/download/1.1.0/firmware-secure-1.1.0.json
- https://github.com/solokeys/solo/releases/download/1.1.0/firmware-hacker-1.1.0.hex

The firmware for the secure build is embedded in a JSON containing a signature that the Solo Secure bootloader verifies before enabling. The private key to generate this signature is our "proprietary secret": this secrecy is what keeps a Solo Secure key safe from "evil maid" attacks, as only firmware approved by us can be flashed onto it. Together with the Solo Secure attestation key, these are the only secrets our company has :open_hands:.

Everything else is 100% verifiable by you, if you wish to do so! Here's how:
- update your checkout of this repository (github.com/solokeys/solo)
- run `make docker-build SOLO_VERSIONISH=1.1.0` (should work on Linux and Windows)
- you will find the following files in the `build/` subdirectory:
```
bootloader-nonverifying-1.1.0.hex
bootloader-nonverifying-1.1.0.sha2
bootloader-verifying-1.1.0.hex
bootloader-verifying-1.1.0.sha2
firmware-hacker-1.1.0.hex
firmware-hacker-1.1.0.sha2
firmware-secure-1.1.0.hex
firmware-secure-1.1.0.sha2
```
- confirm that the assets in https://github.com/solokeys/solo/releases/tag/1.1.0 coincide with your build
- note that the `solo key update` procedure verifies the sha256sums of what will be flashed (including the websafe-encoded version of the secure build `.hex` contained in the signed `.json` asset): https://github.com/solokeys/solo-python/blob/9b5c65db839e28ce5a7a5e526375dc9bc54c259a/solo/cli/update.py#L129

## Bundle Update for Solo Hacker
If you have a hacker key, you can update the bundle of both bootloader + firmware:
- download https://github.com/solokeys/solo/releases/download/1.1.0/bundle-hacker-1.1.0.hex
- alternatively, build your own as above, and run
```
solo mergehex bootloader-nonverifying-1.1.0.hex firmware-hacker-1.1.0.hex bundle-hacker-1.1.0.hex
```
- insert key
- run the following commands:
```
solo program aux enter-bootloader
solo program aux enter-dfu
# powercycle the key (remove and plug in again)
solo program dfu bundle-hacker-1.1.0.hex
solo program aux leave-dfu
# powercycle the key (remove and plug in again)
solo ls
```
In time, the mode changes (between firmware, bootloader and dfu) may be automated, but they can be a little finicky. In particular, when entering or leaving ST DFU mode, a hard reboot (powercycle) can be necessary. Here is a diagram explaining the `solo program` subcommands: https://whimsical.co/YHmbripdnGn9DtBA8YrrdL

## Bug Reports
In case of any issues, please report by commenting in this GitHub issue, so we can fix them

Thank you!

nickray, 2019-02-18:
Fixes thanks to feedback so far:
- `solo key update` did not always detect keys not being in bootloader mode
- on Windows, `solo ls` failed if there is no libusb backend installed

If you encounter these issues, please update to v0.0.5 via `pip install -U solo-python`.

yparitcher, 2019-02-20:
is there a way to convert a Hacker to a Secure Solo, By updating the bootloader etc?

nickray, 2019-02-20:
Nearly. The full Solo Secure (bootloader+firmware) has three differences to Solo Hacker:
- bootloader verifies update signatures
- firmware has ROP level 2 protection enabled
- key contains official attestation secret key

The first two you can simulate, but the last you can't. We can't distribute the analogous `bundle-secure-1.1.0.hex` (consisting of `bootloader-verifying-1.1.0.hex` with self-chosen verifying key and `firmware-secure-1.1.0.json`, with attestation key injected via `solo mergehex --attestation-key <KEY>`) as distributing this with its embedded attestation key would defeat the purpose of attestation. Similarly, the private key for signing updates needs to be kept secret by us, and publishing an "official" verifying private key would defeat the purpose of preventing "evil maid" attacks.

Can you open a separate issue if you'd like to continue discussing this aspect?

DanielNTX, 2019-02-27:
It only worked after I installed the Visual Studio 2015 C++ Build Tools  from here: http://download.microsoft.com/download/5/F/7/5F7ACAEB-8363-451F-9425-68A90F98B238/visualcppbuildtools_full.exe

nickray, 2019-02-27:
@conorpp cut a new release ([v1.1.1](https://github.com/solokeys/solo/releases/tag/1.1.1)) to accommodate a change in behaviour in Chrome 72. The above procedure will update to this version.

DanielNTX, 2019-02-27:
Yeah I noticed my keys went to 1.1.1 when I tested the upgrade.

nickray, 2019-03-14:
Closing this beta program. Both versions 1.1.0 and 1.1.1 had bugs which we believe have been fixed in version 2.0.0. Updates possible via CLI or https://update.solokeys.com.

Thank you to all who took part and reported their issues!

**Please note**: if you took part, are on 1.1.0 and 1.1.1, and have actively used your keys, **you may have to re-register on certain sites** after updating to v2.0.0. So, make sure you have two keys registered, update one by one, re-registering immediately after. Or use backup codes or other fallback methods to make sure you don't lock yourself out anywhere.

closed 2019-03-14 by nickray