solokeys/solo1 #116: Question: Windows 10 Hello

FaganSC, 2019-02-22:
Has anyone been able to set up a SoloKey with Windows 10 Hello?

DanielNTX, 2019-02-26:
Do you already have the AzureAD FIDO2 Authenticator available in your Azure Tenant?

conorpp, 2019-02-26:
Note we still have to implement [hmac-secret](https://github.com/solokeys/solo/issues/106) to support Windows Hello.  I'm not sure if support is possible without it.

Expect an update with hmac-secret/Window Hello support in the coming weeks.

FaganSC, 2019-02-28:
Thanks for the update. Looking forward to trying it out when it's ready!

closed 2019-02-28 by FaganSC

rootext, 2019-07-14:
@conorpp, Hi. Has Windows Hello support been added already? Cannot find this information on the site.

0x0ece, 2019-07-14:
The short answer is we don't know, likely not yet. The good news is that as soon as it's ready, it'll only require a firmware upgrade.

A while back we added support for hmac secret, that in theory should suffice. However we didn't have a way to test Windows Hello as it was a private beta. We asked several people in MS for access, but applications to the program were closed and we didn't have an enterprise customer to support our application. (To date, I still wonder how many people make security keys, and how hard is to add another single beta user.)

In the meantime we heard about other extensions that may be needed, none of which was publicly available or clearly defined (yey for open standards!). Now that Windows Hello is publicly available, a pm reached back to us and we're currently looking into it. I can't tell you yet how much work this will be, hopefully just a couple new extensions, and hopefully we can get the specs soon.

aseigler, 2019-07-14:
I have been testing, it works at least partially. Are there any specific tedt cases you'd like me to try?

rootext, 2019-07-15:
I've bought a Solo key 20 June. With both default firmware and with update to 2.3.0 Windows Hello setup knows that key is inserted but endlessly showing Getting things ready

aseigler, 2019-07-15:
I've successfully registered via AAD (after disabling "attestation" in the preview admin console) , and my tokens replicated into msDS-KeyCredentialLink, but I cannot log on to the machine with Solo. 
[image]

AAD log has similar info as well. Suspect whitelist of trusted roots for "attestation" option in admin page, maybe hmac-secret issue for login?

aseigler, 2019-07-17:
Been looking at this still.  Using a combination of the test tools manifest.exe, cred.exe, and assert.exe and the instructions [here](https://github.com/Yubico/libfido2/tree/master/examples), it seems that there is likely an hmac-secret issue. Specifically, when I use assert.exe with the -s switch to indicate an hmac-secret is requested from the authenticator, the resulting assertion fails signature verification with Solo.  If I omit the -s, the signature is good.