solokeys/solo1 #122: Support for HID gadgets on Linux

rgerganov, 2019-02-27:
I wonder if anyone is interested into using an emulated HID device on Linux which is acting as FIDO2 authenticator. This can be done with the [HID gadget driver](https://www.kernel.org/doc/Documentation/usb/gadget_hid.txt) and could be very useful for doing tests without real hardware. Instead of patching libraries to send HID messages over UDP, one can create a HID device and then read/write to `/dev/hidg0`. I had to patch only `pc/device.c` and it works like a charm.

If anyone is interested into something like this, I may start working on a PR.

andy-dicki, 2019-03-04:
I have compiled it, when I open main, I see the following error:
hidg open: No such file or directory
There is no /dev/hidg*

rgerganov, 2019-03-05:
I am using the following script which creates a HID device (`/dev/hidg0`) on the local machine:
```
MANUFACTURER="Solo"
SERIAL="1234567890"
IDVENDOR="0x0483"
IDPRODUCT="0xa2ca"
PRODUCT="Solo Software Authenticator"
CONFIGFS=/sys/kernel/config

modprobe libcomposite                                             
modprobe dummy_hcd
mount none $CONFIGFS -t configfs                                  
mkdir $CONFIGFS/usb_gadget/fido2
cd $CONFIGFS/usb_gadget/fido2
mkdir configs/c.1
mkdir functions/hid.usb0
echo 0 > functions/hid.usb0/protocol
echo 0 > functions/hid.usb0/subclass
echo 64 > functions/hid.usb0/report_length
echo -ne "\x06\xd0\xf1\x09\x01\xa1\x01\x09\x20\x15\x00\x26\xff\x00\x75\x08\x95\x40\x81\x02\x09\x21\x15\x00\x26\xff\x00\x75\x08\x95\x40\x91\x02\xc0" > functions/hid.usb0/report_desc
mkdir strings/0x409
mkdir configs/c.1/strings/0x409
echo $IDPRODUCT > idProduct
echo $IDVENDOR > idVendor
echo $SERIAL > strings/0x409/serialnumber
echo $MANUFACTURER > strings/0x409/manufacturer
echo $PRODUCT > strings/0x409/product
echo "Configuration 1" > configs/c.1/strings/0x409/configuration
echo 120 > configs/c.1/MaxPower
ln -s functions/hid.usb0 configs/c.1
echo "dummy_udc.0" > UDC
```
Note that you need to have built `libcomposite` and `dummy_hcd` kernel modules.

Will submit the corresponding solo patch for this soon.

andy-dicki, 2019-03-05:
It hanged my computer or disabled all input devices after testing on WebAuthn website. Also, mount command reports that filesystem is already mounted.
Distribution: Parabola GNU/Linux-libre, based on Arch Linux.

rgerganov, 2019-03-05:
> It hanged my computer or disabled all input devices after testing on WebAuthn website. Also, mount command reports that filesystem is already mounted.

Yeah, unfortunately there is kernel bug which causes a deadlock when `usb_f_hid` is used with `dummy_hcd`. I have submitted a [patch](https://marc.info/?l=linux-usb&m=155178064025947&w=2). You can either apply the patch and recompile `usb_f_hid` or use an older kernel (e.g. 4.8.17) which doesn't have this problem.

andy-dicki, 2019-03-05:
Also, I was forced to run Solo as root because otherwise I saw "Permission denied".

rgerganov, 2019-03-05:
> Also, I was forced to run Solo as root because otherwise I saw "Permission denied".

You need to install udev [rules](https://github.com/solokeys/solo/blob/master/udev/70-solokeys-access.rules) for non-root users to be able to access the device

andy-dicki, 2019-03-06:
Does not work on https://fido2.azurewebsites.net/ if "User Verification" is "Required". Is it expected behaviour? Is it same for soft and hardware token?

rgerganov, 2019-03-06:
> Does not work on https://fido2.azurewebsites.net/ if "User Verification" is "Required". Is it expected behaviour? Is it same for soft and hardware token?
> Update: The same on Yubico website if "Passwordless" is enabled.

AFAIK, only Microsoft Edge supports user verification with client PIN at the moment. What browser are you using?

andy-dicki, 2019-03-06:
> AFAIK, only Microsoft Edge supports user verification with client PIN at the moment. What browser are you using?

Tor Browser Bundle, which is based on Mozilla Firefox.

andy-dicki, 2019-03-06:
How can I set PIN anyway?

closed 2019-04-24 by conorpp

alphathegeek, 2019-07-08:
Hi, great thought adding hidg support. I tried this on my Raspberry Pi, and it worked perfectly, in both the testcases, and in actual Use.

I would suggest, that instead of /dev/urandom, for Raspberry Pi's at least, to use /dev/hwrng, since the Pi has a good Hardware random number generator.

PR #131 (2019-04-24): Add support for hidg devices on Linux
There is a HID gadget driver on Linux which provides emulation of USB
HID devices. This could be very useful for testing the Solo firmware
without actual hardware, using only a Linux box.

This patch adds a command line argument which specifies whether the
existing UDP backing should be used or the new one which reads and
writes to /dev/hidg0.

Testing done:
 1. Created HID device with configfs
 2. Started "./main -b hidg" as root
 3. Successfully executed Webauthn registration and authentication on
 the same Linux machine

Closes: #122
commit 955d4f76 (2019-04-24): Add support for hidg devices on Linux

There is a HID gadget driver on Linux which provides emulation of USB
HID devices. This could be very useful for testing the Solo firmware
without actual hardware, using only a Linux box.

This patch adds a command line argument which specifies whether the
existing UDP backing should be used or the new one which reads and
writes to /dev/hidg0.

Testing done:
 1. Created HID device with configfs
 2. Started "./main -b hidg" as root
 3. Successfully executed Webauthn registration and authentication on
 the same Linux machine

Closes: #122