solokeys/solo1 #125: Solokey USB-C stopped working on Gitlab after 1.1.1 firmware update

eccramer, 2019-02-28:
Hey all,

After updating to firmware v1.1.1 today, I tried to log into GitLab with my Solokey. GitLab detects the key, but drops me to a 500 internal server error page instead of logging me in. I logged in last night with the same Solokey, and it worked. I attempted removing and re-adding the Solokey to the account, and the key was detected and added, but Gitlab continues to fail and drop me at a 500 page after attempting to log in.

The updated Solokey continues to work with other accounts that I've registered it with after updating (Google and GitHub log in fine). It also doesn't seem to be a blanket issue with all FIDO2 logins on Gitlab, as I have a basic blue Security Key by Yubikey, which I registered with Gitlab, and logging in with that works fine.

I'm not sure how to get logs from GitLab (this is GitLab.com, not an internal deployment of GitLab), but if someone can point me in the direction of how to extract logs from either the SoloKey, GitLab, or my browser, I can add them to the issue.

I'm using Firefox 65.0.1 and Chrome 72.0.3626.119 on macOS 10.14.3, and my Solokey is a Secure USB C running 1.1.1 firmware.

eccramer, 2019-03-01:
I got home from work and tried logging in with my other Solokey Secure USB-A that I haven’t updated from factory firmware. With the un-updated key on factory firmware, GitLab u2f login works. So I consider this confirmed that v1.1.1 firmware contains a regression on GitLab. 

I’m going to attempt to MitM the handshake between Firefox and GitLab with something like Burp Proxy and see if there are differences beyond the key content.

conorpp, 2019-03-01:
Thanks for catching this!  I haven't had time to look into it, but a couple questions:

Is this with Solo secure?  or with hacker?

Can you try updating with [v1.1.0](https://github.com/solokeys/solo/releases/download/1.1.0/firmware-secure-1.1.0.json)?

```
solo program firmware-secure-1.1.0.json
```

In v1.1.0 I fixed the endianness of the count parameter for U2F.
In v1.1.1 I fixed a return error code for U2F.

eccramer, 2019-03-01:
Hey Connor! No problem! I really like the product and I’m really excited that there’s an open source competitor to Yubico, so I’m happy to contribute anything to help fix issues.

Both keys are Solo Secure. 

I’ll downgrade to 1.1.0 when I get home from dinner and retest.

conorpp, 2019-03-01:
Actually I just went ahead and tried a few things and pretty sure I found the issue.  In `1.1.0` I set the upper byte of the counter to `0xff` so it wouldn't cause issues for people's previous registrations when switching the endianness of it (counter should only go up for each auth).

I tried just setting it to 0x00, compiled/programmed, and it works with Gitlab.  Previously it did not, like what you found.  So Gitlab has issues with counter values at least `0xff000000` it seems.  

I think we should release yet another update, with the counter properly starting at zero, and warn that previous registrations may not work anymore.

I hope you can find more issues :D.  If you're interested, you should check out our testing script:
https://github.com/solokeys/solo/blob/master/tools/ctap_test.py

It could use some improvement.  There's a number of areas in FIDO2 and U2F that need more testing, plus it needs to be connected to our continuous integration.

eccramer, 2019-03-01:
I looked through the commits in the last two releases and was wondering if that was the commit that caused the issue. I'm planning on getting a Solo Hacker soon when the NFC model is offered. I only found out about Solokey after the crowdfunding was over. My friend that introduced me to the project bought a bunch of Solo Hacker NFC so I'll probably buy one off of him as soon as his order is delivered.

I'll take a look at your automated tests. I'm familiar with CI and automated testing, so if I can make some improvements, I'll try to make a PR.

Thanks for the super fast response! I have TOTP set up on GitLab as another second factor, so this is not blocking me from using my account, but the fast response really gives me confidence that Solo is going to become a big player in this space.

MaPePeR, 2019-03-01:
So if gitlab reports a 500 "internal server error"  this is a bug in the gitlab implementation?
I mean: even if your device would behave totally out of spec it should just detect that instead of running into a server error, right?

fooflington, 2019-03-01:
I'm seeing this same issue on a USB-A Secure Key running v1.1.1 (but connected via USB-C) on macOS.

eccramer, 2019-03-01:
@MaPePeR It seems to me that there is an issue here with both GitLab and Solo.

@conorpp correct me if I'm wrong, but the original bug here was that the endianness of the counter was backwards. So instead of counting

```
0x0000001
0x0000002
0x0000003
...
```

It was counting

```
0x1000000
0x2000000
0x3000000
...
```

The upper `0xff` value to the counter was added so that people wouldn't have to reregister their keys after you fixed the endianness issue, as the values reported by the key should always increase. After 1.1.0, the counter started counting 

```
0xff000001
0xff000002
...
```

Which is a really clever solution, but apparently some u2f implementations don't work so well with such big counter values. So really, the counter SHOULD start at `0x00000000`, but after a `0xff000000` logins with a key, GitLab will start failing again.

I think that's reasonable for now. It will take a long time to hit that limit. It's gonna suck reregistering my keys, but I think the mitigation of fixing the underlying counter algorithm is right.

conorpp, 2019-03-01:
Yeah technically this issue is a problem with Gitlab, as there's nothing wrong with having a counter that high.  But it's a bit unexpected and increases the risk for conformance issues.  I'll see about reporting the issue to them as well so at least they can fix it on their side.

I plan to release another update soon, and will add a warning in our pip tool when increasing past `1.1.0` that some registrations may break due to the counter.

mporrato, 2019-03-02:
@conorpp just FYI, I had a very similar issue with gitea after upgrading to 1.1.0: https://github.com/go-gitea/gitea/issues/6168

conorpp, 2019-03-02:
Thanks for sharing.  Never thought having a larger counter would break multiple server implementations,  Will release another update ASAP.

mporrato, 2019-03-02:
I'd say Solokey's behaviour is perfectly aligned to the specs. Implementations rejecting high values of the counter are at fault here, IMHO.

conorpp, 2019-03-02:
@mporrato I agree, but I think it's still good to be as conforming as possible while still in spec.

I just tried register + auth on gitlab with a key with the counter upper byte set to `0x7f` instead of `0xff`.  It seemed to work fine.  Maybe `0x7f` would a good compromise?  Upgrading from 1.1.0 may still require re-registering, but upgrading from before 1.1.0 should be fine.

mporrato, 2019-03-02:
@conorpp that's a brilliant idea, and it should work with gitea as well! Is it possible for the upgrade to do a bitwise or of the MSB of the counter with 0x7f instead of just overwriting it?

0x0ece, 2019-03-02:
The issue seems to be here, and the MSB is a good hint:
https://github.com/castle/ruby-u2f/blob/master/lib/u2f/sign_response.rb#L29

I feel like that function works for ascii chars (i.e. less than 128), but not really on extended ascii. Unless my terminal is hiding something weird:
```
ruby -e "puts 'x0000'.byteslice(1, 4).unpack('N')"
808464432
```
Check the output in hex, you'll see 0x30 repeated.

But:
```
ruby -e "puts 'xðððð'.byteslice(1, 4).unpack('N')"
3283141552
```
In hex this is not a repeated pattern :/

If this is the case/confirmed, the 0x7f will fix, but there might be mess after 256 uses (it won't be the most significant byte, it will be another byte screwing up everything). So probably the best thing is really to push Gitlab (and ruby-u2f) for a fix.

MaPePeR, 2019-03-02:
If that implementation trips so easily about user data that is technically according to the specification i'm worried what happens if you send it out-of-spec-data on purpose...

0x0ece, 2019-03-02:
@MaPePeR You can try... https://hackerone.com/gitlab :)

mporrato, 2019-03-02:
@0x0ece : I had a quick look at the gitlab source code. I'm not familiar with ruby, but the code looks all right to me. I am pretty sure the problem here is the exact same problem I had with gitea: it's not a problem with the application itself but rather with the ORM used by the app, which, for gitlab is ActiveRecord: https://stackoverflow.com/questions/33517403/integer-out-of-range-in-postgresql-database .
Both xorm for gitea and ActiveRecord for gitlab, when used in conjunction with a PostgreSQL database map 32 bit unsigned integer to PostgreSQL's integer type which is a *signed* 32 bit value: since PostgreSQL doesn't have unsigned types, a more acceptable behaviour would be to use bigint instead.

0x0ece, 2019-03-03:
@mporrato oh yes, this makes totally sense to me. We should file a ticket to gitlab to use bigint (or unsigned)

mporrato, 2019-03-03:
I think the solution would be to change https://gitlab.com/fjsanpedro/gitlab-ce/blob/master/db/schema.rb#L1698 to look like `t.integer "counter", limit: 8` instead of `t.integer "counter"`
but I cannot test the change at the moment and I don't have a gitlab account to submit the issue.

webs86, 2019-03-07:
Hi, any update about this? I have the same problem after update my solo key.
I need downgrade to 1.1.0? How can do downgrade? If I use `solo program firmware-secure-1.1.0.json` I get this error:

`Error: No such command "firmware-secure-1.1.0.json".
`

conorpp, 2019-03-08:
New release is out `2.0.0`, shouldn't have any issues with Gitlab.

```
solo key update # --secure or --hacker
```

@webs86 to downgrade, you need to run something like this:
```
solo program bootloader firmware-secure-1.1.0.json
```

jiscfoo, 2019-03-09:
@conorpp I've just tried the online updater and a fresh Ubuntu VM with a fresh `pip install` of the updater and both are reporting 1.1.1 is the latest. Is the 2.x _stable_?

fooflington, 2019-03-09:
@conorpp I've just tried the online updater and a fresh Ubuntu VM with a fresh pip install of the updater and both are reporting 1.1.1 is the latest. Is the 2.x _stable_?

nickray, 2019-03-10:
@fooflington both the online updater and the `solo key update` method always reference the file https://github.com/solokeys/solo/blob/master/STABLE_VERSION to determine the current stable version; you can do so too. As of right now that's 1.1.1.

We're being a little careful before announcing 2.0.0 as stable, although we expect no issues. Why? The fix for the U2F counter endianness issue released in 1.1.0 caused this current issue with GitLab as a side effect. While this side effect is due to an incorrect implementation on GitLab's side, there can always be such externalities for sites and browsers, so we'd like to see if there are any issue reports for adventurous 2.0.0 users before marking it as stable. Basically, we don't want to have everyone update to 2.0.0 before being a little more certain there are no new side effects.

To update to 2.0.0 today, either you can use the manual method @conorpp mentioned above, or you can update `solo` to version 0.0.7 (check via `solo version`) and run:
```
# secure
solo key update --secure --alpha
# hacker
solo key update --hacker --alpha
```
This `--alpha` flag tells `solo` to use the version in https://github.com/solokeys/solo/blob/master/ALPHA_VERSION for the update, which is set to 2.0.0 currently.

If you want or have to do this, be aware that as noted in https://github.com/solokeys/solo/releases/tag/2.0.0, "...if you're upgrading from 1.1.0 or 1.1.1, **AND** you've used your key to authenticate to an services, **then those services may need to be re-registered**. This is due to the counter decreasing."

webs86, 2019-03-10:
@conorpp I have updated my solo key to 2.0.0 and update fix issue with gitlab. I didn't have to re-register my key

fooflington, 2019-03-11:
@nickray @conorpp thanks, apologies – to report state after upgrading to 2.0.0 (should this be a separate issue?):

1. GitLab worked (no reregistration required)
2. Google needed re-registering 
3. GitHub needed re-registering
4. Dropbox worked fine
5. Facebook worked fine

nickray, 2019-03-11:
Thanks @fooflington, that's valuable information. I opened https://github.com/solokeys/solo/issues/140 to collect v2.0.0 feedback.

closed 2019-04-24 by conorpp