solokeys/solo1 #126: independent security audit

shibumi, 2019-03-01:
I strongly suggest an independent security audit for the soft- and hardware.

conorpp, 2019-03-01:
Nitrokey has plans to perform a security audit for the software at some point

shibumi, 2019-03-01:
thanks for the answer and your transparency. What has nitrokey to do with this project?

conorpp, 2019-03-01:
Nitrokey has plans to leverage some of our FIDO2 functionality and will contribute back.  One of their requirements is going through a security audit, so we both get to benefit :)

shibumi, 2019-03-01:
@conorpp nice news!

shibumi, 2019-11-03:
@0x0ece does this mean that solokey got an independend security audit?

0x0ece, 2019-11-03:
No audit and no funds to do one anytime soon.

Fo now FIDO2 L1 certification is the best assurance of security against online attacks.

Independently, we're trying to clean up issues starting from the ones with no activity. This morning we had 75+ issues, it's impossible to keep track of them. I feel issues should give a representation of what's happening and it's being worked on. For ideas and wishlist we should prob find a different place.

MaPePeR, 2019-11-04:
> I feel issues should give a representation of what's happening and it's being worked on. For ideas and wishlist we should prob find a different place.

@0x0ece You could use Issue Labels and Milestones for this.

shibumi, 2019-11-04:
Yes, there are labels and milestones.. I don't think there is a reason to close issues, especially issues asking for an independend security audit.. It looks really suspicious if you close such things without any word.

0x0ece, 2019-11-04:
This is a won’t fix, added label. Suspicious or not, there’s no budget.

shibumi, 2019-11-04:
@0x0ece I don't think you need a budget for this.. I would just leave that issue open and maybe some nice security researcher out there would have an independend look on your hardware.I mean you could at least try to find a security researcher checking your device and if it's with a `We are unhackable`-campaign.. then all the researchers would come for free to just proof you wrong.

wucke13, 2019-11-04:
I second this @shibumi .

shibumi, 2019-11-04:
@0x0ece btw, does this all mean that you are not partnering up with nitrokeys? Because that's it sounds like for me.. @conorpp seemed pretty confident that there will be an independ security audit by nitrokey (they had a few in the past).

jans23, 2019-11-05:
We (Nitrokey) did a short review of the most important security related aspects. For us, the firmware update was most crucial because it affects future firmware releases too and can't be updated itself. One shortcoming we identified and patched subsequently was that old firmware versions could be installed. Now with the patch, only newer firmware versions can be installed (downgrade protection). Because the review wasn't complete we didn't write a review. It would still be good to have a comprehensive security review. At a later stage, we may work more on this.

0x0ece, 2019-11-05:
I reopened then. But I confirm no current budget to do a security review.

shibumi, 2019-11-05:
@jans23 Thanks a lot! This comment is gold for me. So, can we expect a review about this downgrade protection?

0x0ece, 2019-11-05:
Downgrade protection: https://github.com/solokeys/solo/pull/238

This may also be interesting, fault injection in USB: https://github.com/solokeys/solo/pull/224
(physical attack, not really realistic for 2FA, maybe for passwordless but unclear)

jans23, 2019-11-05:
@shibumi We implemented the downgrade protection so somebody else should review it.

0x0ece, 2020-02-19:
Update. We run a security analysis with @Doyensec.

Blog post: https://solokeys.com/blogs/news/security-analysis-of-the-solo-firmware-by-doyensec

Full report: https://doyensec.com/resources/Doyensec_SoloKeys_TestingReport_Q12020_v3.pdf