solokeys/solo1 #130: Solo key emulation

andy-dicki, 2019-03-04:
Solo key emulator (which is detected as Solo key) might be needed for developers who don't have Solo key.

nickray, 2019-03-04:
related: https://github.com/solokeys/solo/issues/122

you could also try to port to a different cortex m4 that has a QEMU, e.g. STM32F429I-Discovery (https://gnu-mcu-eclipse.github.io/qemu/)

conorpp, 2019-03-06:
For development I think it's easiest for now to use the UDP bridge to the PC build.  @andy-dicki have you tried it?

```
cd solo/
make
./main
```

```
python tools/ctap_test.py sim u2f
```

Testing/developing JS APIs could be done with Node.

andy-dicki, 2019-03-07:
Will firmware flashing work for it?

conorpp, 2019-03-07:
I'm not sure what you mean.  It is building the firmware, but for your PC, and then running it directly.  The source is the same for both, expect `device.c` and `crypto.c` are different.

andy-dicki, 2019-03-07:
I mean, _emulate_ device, not protocol.

conorpp, 2019-03-07:
Okay, I just suggest since it can work well for development without a solo, as I often use it.

nickray, 2019-03-07:
@andy-dicki there are no plans to emulate solo device on pc, this is completely out of scope. for this, you might look into QEMU.

conorpp, 2019-04-14:
Closing since a good solution is presented in #131

closed 2019-04-14 by conorpp