solokeys/solo1 #135: Yubico OTP

andy-dicki, 2019-03-07:
Implement Yubico OTP if possible/required.

aveao, 2019-03-08:
There's also static password storage.

sh7dm, 2019-05-07:
Password and GPG/SSH key storage will be very useful

vwbusguy, 2019-06-06:
I would love to see TOTP support similar to Yubico Authenticator.  The key would need to store TOTP registration codes and labels and be able to present that to a client that can generate the tokens.  The simplest way to generate the code for most cases is using oathtool: `oathtool --totp -b <TOTP registration code>`.

trscavo, 2019-06-06:
On Thu, Jun 6, 2019 at 1:08 PM Scott Williams <notifications@github.com> wrote:
>
> I would love to see TOTP support similar to Yubico Authenticator.

OATH TOTP requires a clock, and a clock requires a battery, which
leads to a completely different kind of authenticator. I think there
are other features that are more valuable. Just my two cents.

Tom

evaryont, 2019-06-12:
The existing Yubikeys support TOTP but without a battery -- which is why it only works when paired with something (phone or desktop/laptop running Yubico's apps), and can not generate a TOTP codes alone. (Also, no display!)

I'd also love to see the Solo USB-C come out with TOTP support.

markusC64, 2019-11-25:
I'd like to vote for "HOTP/TOTP (Yubico Authenticator)".

mietzen, 2020-01-14:
I'm voting for HMAC-SHA1