solokeys/solo1 #137: Solo verify gives InvalidSignature

Mirclus, 2019-03-09:
My Solo does not work with any website and `solo key verify`gives an `InvalidSignature` error ([full output](https://gist.github.com/Mirclus/eed99ef1a1c206b11da3daef9f0f763e)). I already tried reseting the keys and all 3 signed firmwares (1.1.0, 1.1.1 and 2.0.0).

This was tested on Arch Linux (Kernel 5.0.0) with Python 3.7.2

More details:

After receiving my Solo Hacker, I started trying some stuff. Unfortunately, I hit bug #50, didn't check the issue tracker and ended up with a locked bootloader (`solo program aux bootloader-version` gives version `1.0.0.`) and a broken firmware.

After leaving it lying around for some time, I tried flashing the signed firmware (1.1.0). The key booted a firmware again, but I had the issues above. I then updated to 1.1.1 and had the same issues. Same story with 2.0.0 today.

Is there any way to recover the Solo or is it gone for good?

nickray, 2019-03-10:
I understand you are using a Solo Hacker. Could it be that you mixed flashing secure and hacker builds?

TLDR: Try running `solo key update --hacker`, and verify again.

In more detail, I suspect two things have happened:
- probably right now you have a secure firmware flashed. What happens during `solo key verify` is the following: Your Solo key is asked to and generates a new credential, adds its certificate, and signs the credential with what it thinks is the certificate key. This key is what distinguishes a Solo Secure from a Solo Hacker; the Solo Secure key is **secret**, exactly for this purpose. When you flash a secure build on a hacker key, the certificate gets swapped out (to the Solo Secure one listed in https://docs.solokeys.io/solo/metadata-statements/), but the certificate key stays the same (it's located somewhere in Flash memory, and set when a key is programmed with a bootlader+firmware bundle via the `solo program dfu` method). In this sense, the hacker Solo key claims it's a secure Solo key now, which it isn't.

- not relevant to your issue, but for your information: flashing a secure build at least once locked the ST DFU bootloader and activated Flash read-out protection level 2, these are both irreversible, so you can no longer do all the things one can usually do with a Solo Hacker. This is not _too_ terrible; you can still flash your own custom firmware by a) activating the bootloader manually (by pressing the button for a few seconds upon insert until the LED flashes), and b) running `solo program bootloader <your_firmware.hex>`.

We should probably handle this case in `solo key verify`. Please let us know if the TLDR fixes your issue.

Mirclus, 2019-03-10:
I tired that already:

```
% solo key update --hacker
We are about to update with the latest Solo Hacker firmware.
Please confirm that the connected Solo key is a Solo Hacker [y/N]: y
Wrote temporary copy of firmware-hacker-1.1.1.hex to /tmp/tmp3d2keutv.hex
sha256sums coincide: 9ed3c4a5dbc848a045be8bcefaef832abc010c9feb3dadcba2a307685e72a845
erasing firmware...
updated firmware 100%             
time: 7.44 s
bootloader is verifying signature...
...error!

Your key did not accept the firmware's signature! Possible reasons:
  * Tried to flash "hacker" firmware on secure key
  * Tried to flash "hacker" firmware on custom hacker key with verifying bootloader

Currently, your key does not work. Please run update again with correct parameters
```

Afterwards the key is stuck in the bootloader (yellow flashing). Only the signed firmwares can be flashed successfully. Every firmware I tried to build myself errors out with `fido2.ctap.CtapError: CTAP error: 0x27 - OPERATION_DENIED`.

conorpp, 2019-03-19:
Can you try the web update?
https://update.solokeys.com/

conorpp, 2019-04-14:
If still not working, perhaps send hello@solokeys.com and link here, and we can send you a new (working) device.

Mirclus, 2019-04-14:
OK, call me confused. I've updated to 2.1.0, and did some more investigating, as the key wasn't accepted by any major provider, nor by the Yubikey test page. So I looked for a page which gave me some debugging output and found https://webauthn.org/.

First, I tested Firefox. It allows me to anonymize the data sent to the server. Interestingly choosing different options here leads to different results (Full outputs at https://gist.github.com/Mirclus/c127f037adef3b30f5235f30ceb028e4):

If I enable anonymization in Firefox, the key seems to work. If I disable it, the key gets rejected with "U2F attestation signature verification failed". In Chrome, I cannot get it to work at all. If I allow access to private data, it gets denied, and if I block this access, the whole process gets aborted.

I'm not sure what information Firefox strips off, but that seems to be the problem. If you need more testing, just say so.

Thank you for the offer, I'll get in touch,

conorpp, 2019-04-14:
I'm thinking the attestation key in the device is incorrect somehow.  I believe the anonymization feature removes the attestation information so it works.  You might be able to fix it by running these commands:

```
wget https://github.com/solokeys/solo/releases/download/2.1.0/bundle-hacker-2.1.0.hex
solo program aux enter-bootloader
solo program aux enter-dfu
solo program dfu bundle-hacker-2.1.0.hex
```

Mirclus, 2019-04-14:
Unfortunately, I can't get into DFU.

ponchick, 2019-07-09:
I have a similar problem. I started with bundle-hacker-2.2.0.hex on NUCLEO-L432KC board. Then I compiled version 2.2.2 and updated firmware. At that point all worked fine. I successfuly added 2nd factor auth to google account, register and auth on demo.yubico.com. Also `solo key verify` worked fine.
Then I compiled locked firmware (make build-release-locked) and flashed `all.hex`. Now I can't authenticate on demo.yubico.com and `solo key verify` produce InvalidSignature error.  Upgrading to 2.3.0 (`solo key update --secure`) doesn't fix the problem. 
Can the problem be fixed by wiping all flash (except bootloader, of course) and reflashing secure firmware? If so, can you provide signed "0xff" dump or small firmware that wipe all certs/keys/counters/etc?