solokeys/solo1 #140: Alpha Program (for firmware v2.0.0 update)

nickray, 2019-03-11:


nickray, 2019-03-11:
Report from @fooflington:

> @nickray @conorpp thanks, apologies – to report state after upgrading to 2.0.0 (should this be a separate issue?):
> 
> GitLab worked (no reregistration required)
> Google needed re-registering
> GitHub needed re-registering
> Dropbox worked fine
> Facebook worked fine

nickray, 2019-03-11:
Report from @webs86:

> @conorpp I have updated my solo key to 2.0.0 and update fix issue with gitlab. I didn't have to re-register my key

cstuder, 2019-03-15:
Updated to 2.0.0 with a Solokey USB-A:

GitHub worked fine
Dropbox worked fine

(Using Chrome 72 on Windows 10)

ukscone, 2019-03-15:
> Updated to 2.0.0 with a Solokey USB-A:
> 
> GitHub worked fine

> 
> (Using Chrome 72 on Windows 10)

i'm being dense and can't find out how to get a solokey working with github. all it seems to offer me is a phone app or sms. Any chance you could write up the steps in words of one syllable so i can under stand it please :)

nickray, 2019-03-15:
You're not dense :) A lot of sites only let you use U2F as 2FA **after** setting up SMS or TOTP first. The thinking is probably that they're scared users will lock themselves out, leading to angry users and/or support requests.

The process for GitHub: https://help.github.com/en/articles/configuring-two-factor-authentication

You can find more deep links to instructions here: https://krypt.co/start/?noredirect=true

conorpp, 2019-04-25:
Thanks for feedback everyone.  Latest firmware is now `2.2.2`.  If you have any more issues, please report.

closed 2019-04-25 by conorpp