solokeys/solo1 #141: Unable to register key

cfasnacht, 2019-03-11:
I'm unable to register my solo keys.
I have two solo keys, with the firmware versions:
1.1.1 and 2.0.0 (which I updated because of the Gitlab Issue)

If I try to register my key with Github the following error appears.
"This device cannot be registered."

I tried with Firefox 65.0.1 and Chromium 72.0.3626.121

conorpp, 2019-03-11:
To confirm, you tried with the 2.0.0 version?  Gitlab works on the same browser + operating system?

I'm not able to replicate on Chrome 72.0.3626.121.

cfasnacht, 2019-03-11:
No both Github and Gitlab don't work, I was never able to register one of the keys successfully.

conorpp, 2019-03-11:
Which firmware file did you use to update?  Hacker or Secure?  If you have a "hacker" edition key, and you update with a secure build, then it will cause this issue

cfasnacht, 2019-03-11:
@conorpp No I don't have the hacker versions and I upgraded one of the keys with: solo key update --secure to 1.1.1 and the other solo key update --secure --alpha to 2.0.0
solo key verify output for both keys:
Please press the button on your Solo key
Register valid
Valid Solo Secure firmware from SoloKeys

conorpp, 2019-03-11:
Does the LED turn orange?  what operating system are you on?

cfasnacht, 2019-03-11:
No the led is green. Ubuntu 18.04.2 LTS

conorpp, 2019-03-11:
Okay, so when you start a register request, it appears nothing happens on the key?

cfasnacht, 2019-03-11:
Yes the led just continues to dim up and down in green

conorpp, 2019-03-11:
So when you run `solo key verify`, you don't press the button?  It just works?

cfasnacht, 2019-03-11:
No I have to press the button otherwise this error is displayed.
Please press the button on your Solo key
Error getting credential, is your key in bootloader mode?
Try: `solo program aux leave-bootloader`

conorpp, 2019-03-11:
Does it work without `sudo`?

conorpp, 2019-03-11:
If you haven't already, please install the udev rules for Linux:

This should work:
https://docs.solokeys.io/solo/udev/

These rules are ideal for everything (reading logs, programming, etc).
https://github.com/solokeys/solo/tree/master/udev

cfasnacht, 2019-03-11:
Thanks for your help with the udev rule it works.

closed 2019-03-11 by cfasnacht