solokeys/solo1 #144: Add udev rules to Linux distributions

nickray, 2019-03-13:
It's too much effort for normal people to install udev rules.
On Debian [1], 70-u2f.rules is packaged with libu2f-host, which is maintained by Yubico.
For Fedora, according to [1], an alternative is used: https://github.com/amluto/u2f-hidraw-policy

[1] https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=824532

nickray, 2019-03-13:
https://github.com/Yubico/libu2f-host/pull/117

nickray, 2019-03-13:
https://github.com/systemd/systemd/issues/11996

Square252, 2019-05-20:
I created PKGBUILDs for ArchLinux (they are available in the AUR):

- https://aur.archlinux.org/packages/solokeys-udev-git/
- https://aur.archlinux.org/packages/solokeys-udev/

emaste, 2019-06-09:
For reference the most recent version of FreeBSD's https://www.freshports.org/security/u2f-devd includes SoloKeys entries
```
        match "vendor"          "0x0483";
        match "product"         "(0xcdab|0xa2ca)";
...
        match "vendor"          "0x1209";
        match "product"         "(0x5070|0x50b0)";
```

nickray, 2019-06-09:
yeah, libu2f-host cut a release, Arch Linux has us now too.

nickray, 2019-07-27:
https://tracker.debian.org/pkg/libu2f-host

Forage, 2019-11-05:
FYI: Ubuntu 19.10 has libu2f-udev 1.1.10-1 installed by default now which includes rules for Solo

poettering, 2020-04-27:
Note that since systemd 244 udev detects fido2 keys automatically and sets the right perms. Any additional udev rules should be unnecessary, in particular any that use vid/pid matching.

jolo1581, 2020-04-27:
@poettering since not everyone hast systems >= 244 the udev rules might be necessary

poettering, 2020-04-27:
My comment was mostly a suggestion to add this to the docs (in particular packaging/building docs), so that people who prepare any of this for distros just skip the rules if their distro has a new systemd.

jolo1581, 2020-04-27:
@poettering 
Maybe you can send a pullrequest

FabianHenneke, 2020-05-08:
@jolo1581 @poettering I added a note about the systemd support in #435.

bjacquin, 2020-05-08:
> @poettering since not everyone has systemd >= 244 the udev rules might be necessary

not everyone has systemd overall