solokeys/solo1 #146: Work once one time on Manjaro

rafaelmr2, 2019-03-18:
Hi,

I configure some services with solokey on Manjaro (github, nextcloud, tutanota). I put the udev files and was working. When I tried again (after reboot), didn't work anymore. 

`udevadm monitor --environment --udev | grep DEVNAME`
DEVNAME=/dev/bus/usb/001/014
DEVNAME=/dev/usb/hiddev0
DEVNAME=/dev/hidraw2
DEVNAME=/dev/bus/usb/001/014

when I insert solokey (the first time was hidraw0). When I press the button nothing happens. Lookin at `dmesg` shows the message below when I press the button:

[ 1993.487134] atkbd serio0: Unknown key pressed (translated set 2, code 0x8d on isa0060/serio0).
[ 1993.487139] atkbd serio0: Use 'setkeycodes e00d <keycode>' to make it known.

I updated to firmware 2.0.0 too and the problem persists.

Can you help me? Thanks for you work!!!

conorpp, 2019-03-18:
What are you expecting to happen when you press the button?  Is there a FIDO request from your browser?

nickray, 2019-03-18:
@rafaelmr2 maybe your udev rule is lacking the leading `SUBSYSTEM=="hidraw"`? I updated https://docs.solokeys.io/solo/udev/ to hopefully clarify. If you think your udev is correct, you can maybe see more info if you run `pip3 install -U solo-python --user` and see what commands like `solo ls`, `solo key wink` and `solo key verify` do.

rafaelmr2, 2019-03-18:
> What are you expecting to happen when you press the button? Is there a FIDO request from your browser?

Yes, when the service request.

rafaelmr2, 2019-03-18:
> @rafaelmr2 maybe your udev rule is lacking the leading `SUBSYSTEM=="hidraw"`? I updated https://docs.solokeys.io/solo/udev/ to hopefully clarify.
> 
> If you think your udev is correct, can you try `pip3 install -U solo-python --user` and see what commands like `solo ls`, `solo key wink` and `solo key verify` do?

It's working with new udev file. 

Thanks!!!

closed 2019-03-18 by nickray