solokeys/solo1 #150: Control and data interfaces are not separated!

nickray, 2019-03-21:
Our CDC-ADC makes Linux complain: https://github.com/torvalds/linux/blob/d92da1fbb72490f999b7d0e809d13d0d52dc78ac/drivers/usb/class/cdc-acm.c#L1234-L1236

It seems we should split out the bulk IN/OUT into a separate interface.

[our-serial.log](https://github.com/solokeys/solo/files/2994313/our-serial.log)
[another-serial.log](https://github.com/solokeys/solo/files/2994314/another-serial.log)

conorpp, 2019-08-12:
Should be fixed here:
https://github.com/solokeys/solo/tree/fix_cdc_interfaces

Darn Windows needs a special non-standard additional descriptor to get it to work with the separated interfaces.

nickray, 2019-08-12:
Windows is the only platform that mixes up the interfaces :package: 
This is how we do it in Rust land: https://github.com/mvirkkunen/usbd-serial/blob/master/src/cdc_acm.rs#L110-L160

conorpp, 2019-08-12:
I dont believe it will