solokeys/solo1 #156: How many FIDO2 resident keys can be stored in a Solo key?

Frederick888, 2019-03-28:
Not sure whether this is the right place for such kind of questions... Allow me to apologise in advance if it isn't, but it would be nice if you guys can save me hours of time digging into source codes.

It is suggested by the document that Solo keys have a 256KB memory but I wonder if I use the key solely for FIDO2, how many resident key credentials can a Solo key hold?

According to https://www.w3.org/TR/webauthn/, a key needs be capable of storing:

field | size
-- | --
id | max 64 bytes
displayName | min 64 bytes
name | min 64 bytes
icon | min 128 bytes

And since Solo adopts key wrapping and `R` is stored in servers (is it stored locally as an identifier of other data?), it'd be 320 bytes in total without taking any overheads into consideration. If Solo keys do not compress the data, it would be able to store ~500 keys I reckon?

0x0ece, 2019-03-28:
We currently support 50 resident keys. To my knowledge yubikey 5 supports 25.

Note that these are the resident ones, “regular” auth doesn’t require resident keys, so you can register Solo on an unlimited number of sites.

Frederick888, 2019-03-28:
@0x0ece Thanks for your reply.

May I know what did you mean by "regular" auth? U2F?

0x0ece, 2019-03-28:
For both U2F and FIDO2, keys are generated "on the fly", so there's no limit on the number of sites. You can read the details here:
https://developers.yubico.com/U2F/Protocol_details/Key_generation.html

Resident keys in FIDO2 are used for password-less + username-less authentication, which is kind of a special case. With "regular" I was referring to the "regular" (lol) case of two-factor authentication, either with u2f or fido2.

nickray, 2019-03-28:
To help your source code dig:
- the resident keys are stored (uncompressed) in 10 dedicated pages (each 2KiB) of the flash memory: https://github.com/solokeys/solo/blob/master/targets/stm32l432/src/memory_layout.h#L23-L26
- "storing/using" a key just copies the key to/from an index of an array in flash memory: https://github.com/solokeys/solo/blob/master/targets/stm32l432/src/device.c#L550

We can't use all of the flash memory for keys, as the code needs to be somewhere too :)

IMHO, the resident key interface isn't so well thought out by FIDO: You can't delete a specific key (nor list all resident keys on the device, which makes some sense from a security perspective if you lose the key), so if you run out of space, you have to reset the Solo key and delete all of the resident keys. (One exception, you can replace the key for a given (relying party, user ID) pair.)

Frederick888, 2019-03-28:
@0x0ece Thanks for the clarification! I thought FIDO2 referred to only password-less username-less but if I understand correctly now, the "single factor username + FIDO2 key pair" and "two factor username + password + FIDO2 key pair" should be preferred unless the client device is something that users cannot easily interact with (e.g. an NFC door lock)?

@nickray I agree. It can't be weirder that the interface doesn't allow users to manage stored resident keys. I just tried out the FIDO2 demo at https://webauthn.me now I don't even know which flow it uses and whether I've wasted a resident key slot 😕

0x0ece, 2019-03-28:
We should prob add a reset option to just wipe resident keys, but not the master secret. I can see people using Solo regularly, and doing tests on resident keys on the various demos online.

(I agree list, single delete shouldn't be available for security reasons -- though it's still possible to enumerate providing rpid.)

0x0ece, 2019-03-28:
@Frederick888 you can use FIDO2 for 2FA or for passwordless, all without resident keys. Then you can use resident keys for password+username-less auth.

Frederick888, 2019-03-28:
Thanks a ton for the explanation, guys. I think this issue can be closed now.

PS: I had a look at https://webauthn.me/js/tutorial.js and it didn't specify PublicKeyCredentialCreationOptions.authenticatorSelection.requireResidentKey (default to false) so I guess it didn't consume any storage in my key... whew 😀

closed 2019-03-28 by Frederick888