solokeys/solo1 #157: Solo Key not detected

WNiels, 2019-03-28:
Hi,

i can't update my Solo Key with "solo-python". It seems not to be detected (pluigged in and flashing green). The command "solo ls" results in ":: Solos", "solo key verify" gets me the following error:
```
raise solo.exceptions.NoSoloFoundError("no Solo found")
solo.exceptions.NoSoloFoundError: no Solo found
```

Thanks,

Niels

conorpp, 2019-03-28:
Is it recognized from the browser?
https://update.solokeys.com/

Which build of Windows are you using?

WNiels, 2019-03-28:
Also not recognized.

Version	10.0.17134 Build 17134

conorpp, 2019-04-14:
Can you find any indication on Windows of the device being recognized?
* Device appears as HID device in Device Manager
* Device appears in [Windows USB View program](https://www.uwe-sieber.de/usbtreeview_e.html).

conorpp, 2019-04-24:
I was recently experimenting on a preview build of Windows and had similar issues.  I found that webauthn on browsers still worked, and running `solo ls` in an administrative shell worked.  Not sure how to get `solo` to work in a normal shell.

planeguy, 2019-08-14:
I dunno if this is the right place for this, but I'm also getting this issue. The key does not show up as an hid device. When I plug it in it shows up as Solo Bootloader 2.3.0 (COM4). I tried uninstalling that and replugging and it just comes back as the bootloader again.

adri66, 2020-04-06:
Manually select the driver to USB Input Device
It's work for me ! 😄 
[image]